AdnanHodzic/displaylink-debian #84: dlm.service active, but xrandrt not seeing USB monitor

KendallWeihe, 2017-07-11:
Successfully installed this service (much appreciated), rebooted system (without USB monitor plugged in), checked `dlm.service` and it was active, plugged monitor in, nothing, checked xrandr, did not exist -- system keeps reading the monitor as a disk.

Some specs:
- Dell Inspiron 5577 
- GTX 1050 4GB vRAM
- 1 HDMI & 3 USB 3.0
- Ubuntu 17.04 Zesty 
- USB to HDMI to DVI 
- one external monitor connected through the HDMI --> trying to connect a second through USB

Some outputs:

_Status:_
```
kendall@kendall-Inspiron-5577:~$ systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-07-11 17:30:53 EDT; 4min 43s ago
 Main PID: 1273 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   Memory: 8.6M
      CPU: 203ms
   CGroup: /system.slice/dlm.service
           └─1273 /opt/displaylink/DisplayLinkManager

Jul 11 17:30:53 kendall-Inspiron-5577 systemd[1]: Starting DisplayLink Manager Service...
Jul 11 17:30:53 kendall-Inspiron-5577 systemd[1]: Started DisplayLink Manager Service.
```

I checked the logs with `dmesg` and here's what exists right after I plug in the USB dongle:
```
[  153.487312] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[  153.507506] usb 2-1: New USB device found, idVendor=1d5c, idProduct=2000
[  153.507510] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  153.509245] usb-storage 2-1:1.3: USB Mass Storage device detected
[  153.509521] scsi host3: usb-storage 2-1:1.3
[  153.627161] usb 1-1: new high-speed USB device number 10 using xhci_hcd
[  153.767426] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101
[  153.767430] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  153.767433] usb 1-1: Product: USB 2.0 Hub
[  153.768531] hub 1-1:1.0: USB hub found
[  153.768561] hub 1-1:1.0: 4 ports detected
[  154.075102] usb 1-1.2: new full-speed USB device number 11 using xhci_hcd
[  154.197380] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=013c
[  154.197384] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  154.197387] usb 1-1.2: Product: USB PnP Sound Device
[  154.197390] usb 1-1.2: Manufacturer: C-Media Electronics Inc.      
[  154.208465] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.3/0003:0D8C:013C.0008/input/input24
[  154.267636] hid-generic 0003:0D8C:013C.0008: input,hidraw6: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-0000:00:14.0-1.2/input3
[  154.516479] scsi 3:0:0:0: Direct-Access                               A001 PQ: 0 ANSI: 4
[  154.517799] sd 3:0:0:0: Attached scsi generic sg0 type 0
[  154.517831] sd 3:0:0:0: [sda] 2047 4096-byte logical blocks: (8.38 MB/8.00 MiB)
[  154.517964] sd 3:0:0:0: [sda] Write Protect is off
[  154.517967] sd 3:0:0:0: [sda] Mode Sense: 03 00 00 00
[  154.518089] sd 3:0:0:0: [sda] No Caching mode page found
[  154.518096] sd 3:0:0:0: [sda] Assuming drive cache: write through
[  154.521114]  sda:
[  154.522127] sd 3:0:0:0: [sda] Attached SCSI disk
```

Like I said, the OS keeps trying to read the monitor as a disk, and [I've seen this as well](https://drive.google.com/file/d/0BwNHOkl6hgfANDZPM1lkQXFKalU/view?usp=sharing)

Xrandr output:
```
kendall@kendall-Inspiron-5577:~$ xrandr 
Screen 0: minimum 8 x 8, current 4480 x 1080, maximum 32767 x 32767
eDP-1-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 193mm
   1920x1080     60.05*+  59.93    48.04  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1280x1024     60.02  
   1440x900      59.89  
   1280x960      60.00  
   1360x768      59.80    59.96  
   1152x864      60.00  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   960x600       60.00  
   960x540       59.99  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   800x512       60.17  
   700x525       59.98  
   640x512       60.02  
   720x450       59.89  
   640x480       60.00    59.94  
   680x384       59.80    59.96  
   576x432       60.06  
   512x384       60.00  
   400x300       60.32    56.34  
   320x240       60.05  
HDMI-1-1 connected 2560x1080+1920+0 (normal left inverted right x axis y axis) 673mm x 284mm
   2560x1080     60.00*+
   1920x1080     60.00    50.00    59.94  
   1920x1080i    60.00    50.00    59.94  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       75.00    60.00    59.94  
   720x400       70.08  
```

Also, I have made sure my listproviders are correct (I think -- this could be where I'm wrong)
```
kendall@kendall-Inspiron-5577:~$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x219 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x44 cap: 0x2, Sink Output crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
```

redjamesg, 2017-07-12:
What does 
`lsusb`
reveal ?

KendallWeihe, 2017-07-12:
So I thought about that (about an hour ago) and was unable to see the DisplayLink ID on the BUS ID list. So I thought, _hey maybe my adapter is bad_. So then I went and found a new adapter, and bam it works! Closing this issue. Thanks :)

closed 2017-07-12 by KendallWeihe