AdnanHodzic/displaylink-debian #85: Debian Stretch 9.1 - Compatible? Green Screen + No Detection of Screen + Other Output:

wethepeopleonline, 2017-07-27:
First off I would like to thank you for your time in porting this driver over to Debian. This is absolutely awesome of you and I would like to say that first and foremost.

I just purchased Pluggable DisplayLink USB 2.0 - USB-VGA-165 (Just standard VGA is fine for me.)

I'm not looking for anything fancy... just a triple screen setup for my workflow in Debian.

I have installed the latest driver that you have committed this project with and it runs and installs fine. However I get the green screen.

Below you can see dlm.service is loaded and enabled on start-up.

`dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-07-26 19:10:09 PDT; 3min 8s ago
  Process: 982 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 989 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─989 /opt/displaylink/DisplayLinkManager

Jul 26 19:10:08 iwannafly systemd[1]: Starting DisplayLink Manager Service...
Jul 26 19:10:09 iwannafly systemd[1]: Started DisplayLink Manager Service.
`

When issuing commands into xrandr ... i'm receiving some strange outputs.

For instance:

- xrandr --listproviders

Output: 

/home/brandon# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x49 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 8 associated providers: 0 name:modesetting (I was receiving an index 1 not found error previous.)

Running a "dmesg" ... I can see that it is detecting the USB Device.

- dmesg

Output:

462.339727] udlfb: DisplayLink USB device /dev/fb1 attached. 1600x1200 resolution. Using 7504K framebuffer memory

...

All packages are up to date and latest... I just built this install just days ago prior to ordering this device after extensive research and finding your GitHub project. (Respect btw!)

... With the data that I have provided; are you able to narrow down the reason that xrandr is only showing 1 provider and ARandR is only showing "VGA-1" and "LVDS-1" ... this is an Alienware M17XR3 Laptop with 1 external + laptop screen and then the green screen USB.

Thank you sir for your time and expertise!

Best Regards and God Bless,

- streetdancer

AdnanHodzic, 2017-08-06:
Thanks for the complements :)

Have you seen: 
[Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) and issues [like this one](https://github.com/AdnanHodzic/displaylink-debian/issues/84)

As this doesn't seem to be an issue with utility but your setup ...

Sorsnce, 2017-11-21:
If you are getting a green screen via Debian 9 check your udl.conf. Run the following command and reply back with the results.

cat /etc/modules-load.d/udl.conf

I am successfully running 4 monitors with the Displaylink USB to VGA

AdnanHodzic, 2017-12-25:
Closing due to inactivity.

closed 2017-12-25 by AdnanHodzic

gpillot, 2018-10-27:
I have a similar problem with Debian 9. I had no problem installing the driver but after the reboot, my monitor displays a green screen. Gnome doesn't detect my monitor. The dlm.service is activated: ● dlm.service - DisplayLink Manager Service Loaded: loaded  (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)  Active: active (running) since Sat 2018-10-27 17:10:21 EDT; 23min ago  Process: 688 ExecStartPre=/bin/sh -c modprobe evdi \|\| (dkms install  evdi/4.4.24 && modprobe evdi) (code=exited, status=0/SUCCESS)  Main PID: 696 (DisplayLinkMana) Tasks: 11 (limit: 4915) Memory: 9.5M  CPU: 2.860s CGroup: /system.slice/dlm.service └─696  /opt/displaylink/DisplayLinkManager dmesg \| grep Display displays the following output: [    2.819374] usb 3-2: Manufacturer: DisplayLink [    14.267456] udlfb: DisplayLink Plugable UGA-165 - serial #619411 [    15.163637] udlfb: DisplayLink USB device /dev/fb1 attached. 1920x1080  resolution. Using 8104K framebuffer memory
--

gpillot, 2018-10-27:
I have a similar problem with Debian 9.

I had no problem installing the driver but after the reboot, my monitor displays a green screen.

Gnome doesn't detect my monitor.

The dlm.service is activated:

`● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-10-27 17:10:21 EDT; 23min ago
  Process: 688 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 696 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   Memory: 9.5M
      CPU: 2.860s
   CGroup: /system.slice/dlm.service
           └─696 /opt/displaylink/DisplayLinkManager`

`dmesg | grep Display` displays the following output:

`[    2.819374] usb 3-2: Manufacturer: DisplayLink
[   14.267456] udlfb: DisplayLink Plugable UGA-165 - serial #619411
[   15.163637] udlfb: DisplayLink USB device /dev/fb1 attached. 1920x1080 resolution. Using 8104K framebuffer memory`