AdnanHodzic/displaylink-debian #88: Debian 9 - Green Screen

brunocollaco, 2017-08-18:
Heya, first of all, thanks for sharing this plugin.

I never succeed installing this driver on debian.

Already followed the post installation guide also. The dlm service is active and I can see 2 providers, but my screen keeps showing green, an ugly green btw (=

dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor pres
   Active: active (running) since Thu 2017-08-17 22:11:38 -03; 2min 4s a
  Process: 586 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install ev
 Main PID: 590 (DisplayLinkMana)
    Tasks: 13 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─590 /opt/displaylink/DisplayLinkManager

xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0x120 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

ARANDR shows only my default display.

I appreciate if someone can help. Cheers!

redjamesg, 2017-08-18:
Have you issued the command
[Code]
xrandr --setprovideroutputsource 1 0
[/code]
Before using arandr?

adangel, 2017-08-18:
I see, both providers use the "modesetting" x driver. Maybe it's the same issue describe [in #68](https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674)?

brunocollaco, 2017-08-19:
I've done your suggestions but without success.

xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel
Provider 1: id: 0x145 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Now provider 0 is showing Intel as you described oi #68, but my screen is still green.

)=

Thanks a lot anyway.

benpturner, 2017-09-03:
I am having the same issue, any update on this. Its a black screen but the mouse cursor moves over it so its working. I did supply the xrandr --setprovideroutputsource 1 0 then xrandr --output DVI-I-1-1 --auto --right-of eDP-1

Sorsnce, 2017-11-21:
If you are getting a green screen via Debian 9 check your udl.conf. Run the following command and reply back with the results.

cat /etc/modules-load.d/udl.conf

AdnanHodzic, 2017-12-25:
Closing issue due to inactivity.

closed 2017-12-25 by AdnanHodzic

AdnanHodzic, 2018-12-29:
JFYI as part of [last commit](https://github.com/AdnanHodzic/displaylink-debian/commit/c5a69a97f73413e01ed749d112ec0f12494e7eb6) I've added what I believe is proper xorg support for Intel Corporation HD Graphics 520, which might've been the problem here.