AdnanHodzic/displaylink-debian #89: ./displaylink-debian.sh will not install properly

Manateeoncruise, 2017-08-19:
Trying to run displaylink-debian.sh gives me the following errors:
`jan@jan-Dell-System-XPS-L322X:~/Downloads$ sudo sh displaylink-debian.sh
displaylink-debian.sh: 8: displaylink-debian.sh: Syntax error: newline unexpected
jan@jan-Dell-System-XPS-L322X:~/Downloads$ sudo /bin/bash displaylink-debian.sh
displaylink-debian.sh: Zeile 7: Syntaxfehler beim unerwarteten Wort »newline«
displaylink-debian.sh: Zeile 7: `<!DOCTYPE html>'`
Any help?

redjamesg, 2017-10-24:
Sorry a mistype in the reference made my commit appear under this issue, it has nothign to do with the reported issue here.

redjamesg, 2017-10-24:
Have you tried to run the script without the 'sh' or '/bin/bash'  in the sudo command?
Like:
`sudo displaylink-debian.sh`

If the script itself has the execute bit set, there's no need for the added interpreter, the first line in the script takes care of that.

AdnanHodzic, 2017-12-25:
Closing issue due to inactivity.

closed 2017-12-25 by AdnanHodzic