AdnanHodzic/displaylink-debian #90: No xrandr --listproviders

panosd, 2017-09-10:
Hello, i have installed with no problem displaylink driver on debian 8 but i can't see provider for displaylink except my primary. any help?

```
# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x56 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:ARUBA @ pci:0000:00:01.0
```

```
dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-09-11 00:22:28 EEST; 26min ago
  Process: 610 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && modprobe evdi) (code=
 Main PID: 618 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─618 /opt/displaylink/DisplayLinkManager

Sep 11 00:22:28 themachine systemd[1]: Starting DisplayLink Manager Service...
Sep 11 00:22:28 themachine systemd[1]: Started DisplayLink Manager Service.

```

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected.

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic