AdnanHodzic/displaylink-debian #93: Closing session  and  reissue of session.

rafaelgov95, 2017-10-15:
When I switch to monitor a screen closes and open a new clean session, but nothing mon monitor 2 appears.

AdnanHodzic, 2017-12-25:
Could you please be more elaborate in what excatly the problem is?

edclement, 2018-01-05:
Something similar was occurring for me even prior to installing this driver. These bug reports might be useful. Had to turn off the black screen feature.
* https://displaylink.org/forum/showthread.php?t=65647
* https://displaylink.org/forum/showthread.php?t=65588

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic