AdnanHodzic/displaylink-debian #94: display not detected

laurentL, 2017-10-16:
hi,
I am using a D3100 on a inspiron 5579 under debian buster

[  400.652091] evdi: [D] add_store:239 Increasing device count to 4
[  400.652217] evdi: [D] evdi_crtc_init:351 drm_crtc_init: 0
[  400.652297] evdi: [W] evdi_painter_crtc_state_notify:484 Painter does not exist!
[  400.652299] evdi: [D] evdi_detect:78 Painter is disconnected
[  400.652302] [drm] Cannot find any crtc or sizes - going 1024x768
[  400.653165] evdi evdi.3: fb4: evdidrmfb frame buffer device
[  400.653166] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  400.653166] [drm] No driver support for vblank timestamp query.
[  400.653217] [drm] Initialized evdi 1.5.0 20170419 for evdi.3 on minor 4
[  400.653218] [drm] evdi: evdi_stats_init

 xrandr --listproviders
Providers: number : 0

● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-10-16 05:07:28 CEST; 2min 45s ago
  Process: 6394 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.4.210 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 6397 (DisplayLinkMana)
    Tasks: 35 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─6397 /opt/displaylink/DisplayLinkManager

Oct 16 05:07:28 xxxxxxxxxxxxxx systemd[1]: Starting DisplayLink Manager Service...
Oct 16 05:07:28 xxxxxxxxxxxxxx systemd[1]: Started DisplayLink Manager Service.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected.

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic