AdnanHodzic/displaylink-debian #95: frozen tty on displaylink no x on debian buster

xevilstar, 2017-10-23:
Please I pray you help,

I got a frozen terminal window with the first two lines of bootup on the displaylink monitor and no X.
All other monitors are working fine

find my xorg.conf attached.

some details:

root@azrael:~# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:Intel
root@azrael:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 17e9:0360 DisplayLink 
Bus 001 Device 006: ID 045e:0810 Microsoft Corp. 
Bus 001 Device 005: ID 046d:0a45 Logitech, Inc. 
Bus 001 Device 004: ID 1c4f:0055 SiGma Micro 
[xorg.conf.txt](https://github.com/AdnanHodzic/displaylink-debian/files/1407561/xorg.conf.txt)

Bus 001 Device 003: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@azrael:~# ps -ax|grep Display
  856 ?        Ssl    0:02 /opt/displaylink/DisplayLinkManager
 4317 pts/60   S+     0:00 grep Display
root@azrael:~# uname -a
Linux azrael 4.13.0-1-amd64 #1 SMP Debian 4.13.4-2 (2017-10-15) x86_64 GNU/Linux

xevilstar, 2017-10-23:
[xorg.conf.txt](https://github.com/AdnanHodzic/displaylink-debian/files/1407567/xorg.conf.txt)

xevilstar, 2017-10-23:
https://github.com/AdnanHodzic/displaylink-debian/issues/90

redjamesg, 2018-01-05:
It seems you are missing the identification of the providers.
Have you looked into the post-install help file, I think the mentioned #68 might be of help to you.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected.

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic