AdnanHodzic/displaylink-debian #97: My asus mb169b+ is still black after installing displaylink-debian.

lks21c, 2017-11-03:
I think this would be displaylink with intel gpu issue

OS : ubuntu 14.04.5 LTS(x64)
GPU : intel graphics
DisplayLink Driver : 1.4
Device : Asus MB169B+

I tried to install display link driver correctly followed by the link http://support.displaylink.com/knowledgebase/articles/684649-how-to-install-displaylink-software-on-ubuntu and it didn't work out, so I treid this displaylink-debian.

Also, I could find the device by the command "lsusb -d 17e9:"
which I think the OS detects the monitor via usb 3.0.

But My DL device stays black screen but I could see power button led is orange.
(which means cable connection worked out even though screen didn't turn on)

I tried to detect the device by clicking "detect monitor" on settings but didn't work out.

So I attached dl support log, so could you help me out with that?
attachement : http://www.kwangsiklee.com/wp-content/uploads/2017/11/dl_log.zip

Thanks you a lot in advance!

redjamesg, 2017-11-03:
Have you followed the suggestions in the post-install file?
Especially #15 or #68

lks21c, 2017-11-03:
Thanks a lot. It was caused by [#15 ](https://github.com/AdnanHodzic/displaylink-debian/issues/15) and my device worked out after disable UEFI boot options on CMOS.

closed 2017-11-03 by lks21c