AdnanHodzic/displaylink-debian #99: Display provider not found

itsalme, 2017-11-19:
Hi Adnan,
I am running Debian 9.1 and your latest build (38b4ba5).
displaylink_debian.sh installed driver with no problem and output to screen is possible but only with  proceeding (1). Proceedings (2) and (3) result in a status that only one provider listed, thus the displaylink provider is not available. Restart of the dlm.service does not help to fix the problem. Enclosed the dmesg outputs.

Do you have an idea what's going on ?
Regards Alexander
[1_display_link_ok.log](https://github.com/AdnanHodzic/displaylink-debian/files/1485537/1_display_link_ok.log)
[2_display_link_usb_stays_connected.log](https://github.com/AdnanHodzic/displaylink-debian/files/1485538/2_display_link_usb_stays_connected.log)
[3_display_link_connect_usb_before_login.log](https://github.com/AdnanHodzic/displaylink-debian/files/1485539/3_display_link_connect_usb_before_login.log)

(1) displaylink device is connected after gdm3 login. Displaylink works after chosing the proper provider and display setting. Info: gnome-control-center display lists the two displays only after "xrandr -- setprovideroutputsource"

(2) displaylink device is connect before boot. Only one provider is listed. Dlm.service restart does not help.

(3) displaylink device is connected before gdm3 login. Again only one provider is listed. Dlm.service restart does not help.

AdnanHodzic, 2017-12-25:
@itsalme can you please confirm if this problem is still present with latest version of the driver?

gregchi, 2018-01-06:
I have very similiar issue. It worked only once when I've followed strictly the post installation guide with disconnected screen from the docking station, after connection monitor back another provider appeared and I was able to configure and use it. Worked fine.But since that first time it does not work. I am on Debian GNU/Linux 9.3 (stretch). I used displaylink_debian.sh to install, all modules suppose to be good, services are running..

gregchi, 2018-01-15:
Ok, just figured what works on my debian 9.3...
After login open terminal, go root and:
1. Stop dlm service: 
# systemctl stop dlm.service
2. Unload evdi module:
# rmmod -f evdi
3.Start dlm service (it will load evdi module back):
# systemctl start dlm.service
Now after command: xrandr --listproviders you should see two providers in similiar way to this:

"Providers: number : 2
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 1 name:Intel
Provider 1: id: 0x104 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting"

Then you can do this:
# xrandr --setprovideroutputsource 1 0
and use arandr to set correct orientation of second screen :)

itsalme, 2018-01-18:
Hi Adnan,
the problem still persists with the latest version of the displaylink driver and your script.

itsalme, 2018-01-18:
Hi Gregchi,
your steps help to make "xrand --listproviders" display the displaylink provider. But the "rmmod -f evdi" results in a freeze of the server shutdown process in my setup.

To work around the problem, I adjusted the /opt/displaylink/udev.sh and removed the start of the dlm service in this script. I added the dlm service start into the script to adjust the screen settings within the user context.
In my oppinion the problem boils down, that the evdi kernel module has to be loaded within the user context. The root cause is still not clear to me.
In my setup a problem remains ... logout and login as a different user pushes the system in the problem again.
In my oppinion, it would be helpful to find a way to remove the evdi kernel module without using the "force" option.

Regards Alexander

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic

kir0ul, 2019-02-12:
Hello,

I think I have the same problem on my Debian 9.

If I do `git pull && git checkout modesetting && sudo ./displaylink-debian.sh`, I get:
```
Preparing for install

Archive:  DisplayLink_Ubuntu_4.4.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of DisplayLink_Ubuntu_4.4.zip or
        DisplayLink_Ubuntu_4.4.zip.zip, and cannot find DisplayLink_Ubuntu_4.4.zip.ZIP, period.
chmod: cannot access '4.4/displaylink-driver-4.4.[0-9]*.run': No such file or directory
./displaylink-debian.sh: line 246: ./4.4/displaylink-driver-4.4.[0-9]*.run: No such file or directory
mv: cannot stat 'displaylink-driver-4.4.[0-9]*/': No such file or directory
sed: can't read 4.4/displaylink-driver-4.4/displaylink-installer.sh: No such file or directory

## NO suitable EVDI source found ##

sed: can't read 4.4/displaylink-driver-4.4/displaylink-installer.sh: No such file or directory

-------------------------------------------------------------------

Installing driver version: 4.4

./displaylink-debian.sh: line 282: cd: 4.4/displaylink-driver-4.4: No such file or directory

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory
Disabling PageFlip for modesetting
```

On the `master` branch, I get the following message when I run `displaylink-debian.sh`:
```
Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf
```

It then eventually works if I unload the EVDI module and start back the `dlm` service as written here: https://github.com/AdnanHodzic/displaylink-debian/issues/99#issuecomment-357771602

AdnanHodzic, 2019-02-16:
Please don't use any other branch then `master`, all the changes are merged in it. 

Pull the latest changes in `master` and then try uninstall/install.