AdnanHodzic/displaylink-debian #101: Displaylink lags/low refresh rate

Sorsnce, 2017-11-22:
I have successfully installed and configured the Displaylink to work with Kali 2017.3 the only issue I am having is when the monitor that is using the Displaylink is active my entire systems slows down/really bad refresh rate. I have found if I change my background to completely black it help speed up my system, but its still not running like it should. Has anyone seen this issue using the Displaylink?
Thank you!

zlowram, 2017-11-27:
Yep, same here. It is specially laggy when I just have the external monitor active, and the laptop screen disabled. It's pretty annoying... However I haven't found a solution for it.

Hope someone sheds a bit of light on this!

AdnanHodzic, 2017-12-25:
Think this is a problem related to the driver itself, please try reporting this same issue on: https://github.com/DisplayLink/evdi/issues

Kaffiend, 2018-03-14:
Its related to pageflipping in the driver itself and its falling back to CPU rendering

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic

zlowram, 2018-03-26:
Thanks a lot for this @Kaffiend and @AdnanHodzic!

It's definitely working better, still with some small lagging "glitches" from time to time, but now it is perfectly usable :+1: