AdnanHodzic/displaylink-debian #103: Deepin is also based on debin？Please help us to support?

ricardo0624, 2017-12-21:
Thanks for your great job!
Can you help us to support deepin (another debin based linux)?

AdnanHodzic, 2017-12-21:
Hey @ricardo0624 glad to report that with latest commit displaylink-debian now also supports Deepin.

Enjoy, and let me know if you have any additional questions and/or comments.

ricardo0624, 2017-12-22:
Thanks a lot! mine is 15.5 ,I installed it successfully,but It doesn't work ,case it is 15.5, not support?

AdnanHodzic, 2017-12-25:
What doesn't work? Have you tried looking at: https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md

ricardo0624, 2017-12-26:
@AdnanHodzic I am not familiar with linux, I can't describe it.
I think It completely installs successfully,maybe It is deepin's problem,thanks for your help,I have feedback it to deepin org.
I will try it your help.
Thanks for your help,maybe you should close this issue.I thought it's absolute not your fault.
Thanks  again.

closed 2017-12-26 by ricardo0624