AdnanHodzic/displaylink-debian #104: USB not working until I unplug and plug again.

gsfakianakis, 2017-12-28:
So far I managed to connect screens.
I should stop the dlm service, start again and the connect.

However, mouse and keyboard do not work until I unplug them and plug again.

Is this known issue ? Is there any elegant way to avoid doing this all the time ?

redjamesg, 2017-12-28:
It is not an issue I have noticed, what display link device are you connecting?

gsfakianakis, 2017-12-28:
Kensington USB 3.0 Docking Station (sd3500v) in Debian 9.

redjamesg, 2017-12-28:
When you connect it, what does dmesg show as opposed to when you disconnect/reconnect the keyboard/mouse?
What is the lsusb output from either situations as well?
When you connect the Kensington it should display a USB hub recognised, then within 3 -4 seconds it should identify the mouse/keyboard devices as well.

gsfakianakis, 2017-12-28:
Before connecting the displaylink

----------------------------------------------------------------------
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 6, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 2: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M

----------------------------------------------------------------------

-----------
after connecting (keyboard and mouse do not work)

/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 3: Dev 58, If 0, Class=Hub, Driver=hub/4p, 5000M
            |__ Port 1: Dev 59, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 59, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 59, If 5, Class=Communications, Driver=cdc_ncm, 5000M
            |__ Port 1: Dev 59, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 59, If 1, Class=Application Specific Interface, Driver=, 5000M
            |__ Port 1: Dev 59, If 6, Class=CDC Data, Driver=cdc_ncm, 5000M
            |__ Port 1: Dev 59, If 4, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 3: Dev 60, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 6, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 2: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 3: Dev 123, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 124, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M

----------------------------------------------------------------------
----re connect keyboard mouse by hand

/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 3: Dev 58, If 0, Class=Hub, Driver=hub/4p, 5000M
            |__ Port 1: Dev 59, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 59, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 59, If 5, Class=Communications, Driver=cdc_ncm, 5000M
            |__ Port 1: Dev 59, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 59, If 1, Class=Application Specific Interface, Driver=, 5000M
            |__ Port 1: Dev 59, If 6, Class=CDC Data, Driver=cdc_ncm, 5000M
            |__ Port 1: Dev 59, If 4, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 3: Dev 60, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 6, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 2: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 3: Dev 123, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 13, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 3: Dev 16, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
                |__ Port 1: Dev 14, If 2, Class=Audio, Driver=snd-usb-audio, 480M
                |__ Port 1: Dev 14, If 0, Class=Video, Driver=uvcvideo, 480M
                |__ Port 1: Dev 14, If 3, Class=Audio, Driver=snd-usb-audio, 480M
                |__ Port 1: Dev 14, If 1, Class=Video, Driver=uvcvideo, 480M
                |__ Port 2: Dev 15, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
                |__ Port 2: Dev 15, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M

gsfakianakis, 2017-12-28:
dmesg

Connected and then disconnect display lnk:
[124503.233580] usb 1-3.3.2.3: USB disconnect, device number 16
[124503.274141] usb 1-3.3.3: USB disconnect, device number 125
[124503.274379] usb 1-3.3.4: USB disconnect, device number 127
[124503.276978] evdi: [D] evdi_painter_disconnect:611 (dev=1) Disconnected from ffff9961ebad7000
[124503.276980] evdi: [D] evdi_detect:78 Painter is disconnected
[124503.277007] evdi: [W] evdi_painter_disconnect:593 (dev=-1) An unknown connection to ffff9961ebad7000 tries to close us
[124503.277009] evdi: [W] evdi_painter_disconnect:594  - ignoring
[124503.281939] evdi: [D] evdi_painter_disconnect:611 (dev=2) Disconnected from ffff9960aebc4400
[124503.281941] evdi: [D] evdi_detect:78 Painter is disconnected
[124503.281968] evdi: [W] evdi_painter_disconnect:593 (dev=-1) An unknown connection to ffff9960aebc4400 tries to close us
[124503.281969] evdi: [W] evdi_painter_disconnect:594  - ignoring
[124503.285069] evdi: [D] evdi_detect:78 Painter is disconnected
[124503.285625] evdi: [D] evdi_detect:78 Painter is disconnected
[124503.315250] evdi: [D] evdi_painter_crtc_state_notify:481 (dev=-1) Notifying crtc state: 3
[124503.315252] evdi: [W] evdi_painter_send_crtc_state:387 Painter is not connected!
[124503.315924] evdi: [D] evdi_painter_crtc_state_notify:481 (dev=-1) Notifying crtc state: 3
[124503.315925] evdi: [W] evdi_painter_send_crtc_state:387 Painter is not connected!
[124503.703915] evdi: [D] evdi_detect:78 Painter is disconnected
[124503.703925] evdi: [D] evdi_detect:78 Painter is disconnected

-------------------------

[124520.141623] usb 1-3.3: new high-speed USB device number 17 using xhci_hcd
[124520.243947] usb 1-3.3: New USB device found, idVendor=2109, idProduct=2811
[124520.243949] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[124520.243951] usb 1-3.3: Product: USB2.0 Hub             
[124520.243952] usb 1-3.3: Manufacturer: VIA Labs, Inc.         
[124520.244644] hub 1-3.3:1.0: USB hub found
[124520.244791] hub 1-3.3:1.0: 4 ports detected
[124520.321905] usb 2-3.3: new SuperSpeed USB device number 62 using xhci_hcd
[124520.581148] usb 2-3.3: New USB device found, idVendor=2109, idProduct=8110
[124520.581151] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[124520.581152] usb 2-3.3: Product: USB3.0 Hub             
[124520.581153] usb 2-3.3: Manufacturer: VIA Labs, Inc.         
[124520.582919] hub 2-3.3:1.0: USB hub found
[124520.583119] hub 2-3.3:1.0: 4 ports detected
[124520.669705] usb 1-3.3.2: new high-speed USB device number 18 using xhci_hcd
[124520.783044] usb 1-3.3.2: New USB device found, idVendor=05e3, idProduct=0608
[124520.783046] usb 1-3.3.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[124520.783047] usb 1-3.3.2: Product: USB2.0 Hub
[124520.783832] hub 1-3.3.2:1.0: USB hub found
[124520.784106] hub 1-3.3.2:1.0: 4 ports detected
[124520.881714] usb 1-3.3.3: new high-speed USB device number 19 using xhci_hcd
[124521.429926] usb 2-3.3.1: new SuperSpeed USB device number 63 using xhci_hcd
[124521.450724] usb 2-3.3.1: New USB device found, idVendor=17e9, idProduct=430f
[124521.450728] usb 2-3.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[124521.450730] usb 2-3.3.1: Product: Kensington USB3.0 2X Video Dock
[124521.450732] usb 2-3.3.1: Manufacturer: DisplayLink
[124521.450734] usb 2-3.3.1: SerialNumber: 10073726
[124521.462029] usb 2-3.3.1: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
[124521.462032] usb 2-3.3.1: [15] FU [DisplayLink Audio Playback Volume] ch = 6, val = -8176/0/16
[124521.462609] usb 2-3.3.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
[124521.462610] usb 2-3.3.1: [12] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
[124521.477581] cdc_ncm 2-3.3.1:1.5: MAC-Address: 00:50:b6:c4:9a:12
[124521.477584] cdc_ncm 2-3.3.1:1.5: setting rx_max = 16384
[124521.477718] cdc_ncm 2-3.3.1:1.5: setting tx_max = 16384
[124521.478039] cdc_ncm 2-3.3.1:1.5 usb0: register 'cdc_ncm' at usb-0000:00:14.0-3.3.1, CDC NCM, 00:50:b6:c4:9a:12
[124521.533744] usb 1-3.3.2.1: new high-speed USB device number 20 using xhci_hcd
[124521.638181] usb 1-3.3.3: New USB device found, idVendor=05e3, idProduct=0610
[124521.638184] usb 1-3.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[124521.638185] usb 1-3.3.3: Product: USB2.0 Hub
[124521.638186] usb 1-3.3.3: Manufacturer: GenesysLogic
[124521.639821] hub 1-3.3.3:1.0: USB hub found
[124521.641397] hub 1-3.3.3:1.0: 4 ports detected
[124521.650223] usb 1-3.3.2.1: New USB device found, idVendor=045e, idProduct=0779
[124521.650225] usb 1-3.3.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[124521.650227] usb 1-3.3.2.1: Product: Microsoft® LifeCam HD-3000
[... 1442 lines cut ...]
[124680.169756] usb 1-3.3.2.1: New USB device found, idVendor=045e, idProduct=0779
[124680.169760] usb 1-3.3.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[124680.169761] usb 1-3.3.2.1: Product: Microsoft® LifeCam HD-3000
[124680.169763] usb 1-3.3.2.1: Manufacturer: Microsoft
[124680.171372] uvcvideo: Found UVC 1.00 device Microsoft® LifeCam HD-3000 (045e:0779)
[124680.176708] uvcvideo 1-3.3.2.1:1.0: Entity type for entity Extension 5 was not initialized!
[124680.176712] uvcvideo 1-3.3.2.1:1.0: Entity type for entity Processing 4 was not initialized!
[124680.176714] uvcvideo 1-3.3.2.1:1.0: Entity type for entity Selector 3 was not initialized!
[... cut ...]
[124680.803190] usb 1-3.3.2.3: New USB device found, idVendor=1bcf, idProduct=0005
[124680.803194] usb 1-3.3.2.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[124680.803196] usb 1-3.3.2.3: Product: USB Optical Mouse
[124680.809992] input: USB Optical Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3.2/1-3.3.2.3/1-3.3.2.3:1.0/0003:1BCF:0005.0057/input/input189
[124680.810297] hid-generic 0003:1BCF:0005.0057: input,hidraw7: USB HID v1.10 Mouse [USB Optical Mouse] on usb-0000:00:14.0-3.3.2.3/input0

gsfakianakis, 2017-12-28:
What I can see is that i get this:
"Not enough bandwidth for new device state."

Then it sees me disconnecting:

USB disconnect, device number 18
[124674.808065] usb 1-3.3.2.1: USB disconnect, device number 20
[124674.838251] usb 1-3.3.2.2: USB disconnect, device number 22
[124674.838516] usb 1-3.3.2.3: USB disconnect, device number 23

and after connection everything is fine:

[124680.176933] input: Microsoft® LifeCam HD-3000 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3.2/1-3.3.2.1/1-3.3.2.1:1.0/input/input186

[124680.491494] input: Microsoft Wired Keyboard 600 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3.2/1-3.3.2.2/1-3.3.2.2:1.0/0003:045E:07F8.0055/input/input187
[
[124680.809992] input: USB Optical Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3.2/1-3.3.2.3/1-3.3.2.3:1.0/0003:1BCF:0005.0057/input/input189

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which could resolve your problem.

Could you please run re-install with latest changes and report if this same problem exists?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic