AdnanHodzic/displaylink-debian #106: Hello?KDE NEON is based ubuntu16.04,help us to support it!

ricardo0624, 2018-01-02:
hello,I have installed KDE Neon,it's desktop env is KDE, and it's really great,just 400M memory,gnome is about 1.2G.
So I finally found a great branch.
Thanks you.

AdnanHodzic, 2018-03-24:
KDE Neon should be supported out of box, are you saying this is not the case?

Could you please give me the output of `lsb_release -ics | sed '$!s/$/ /' | tr -d '\n'`

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected.

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

Also, could you tell me what does running `lsb_release -ics | sed '$!s/$/ /' | tr -d '\n'` return for KDE Neon.

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic