AdnanHodzic/displaylink-debian #107: version pattern matching broken in latest driver package (4.1)

debalance, 2018-01-02:
Hi,
it seems the latest driver package (4.1) released on December 21st 2017 has a slightly different file structure / versioning than earlier ones.

>   inflating: 4.1/displaylink-driver-4.1.9.run  
chmod: cannot access '4.1/displaylink-driver-4.1.???.run': No such file or directory
./displaylink-debian.sh: line 245: ./4.1/displaylink-driver-4.1.???.run: No such file or directory
mv: cannot stat 'displaylink-driver-4.1.???/': No such file or directory
sed: can't read 4.1/displaylink-driver-4.1/displaylink-installer.sh: No such file or directory
sed: can't read 4.1/displaylink-driver-4.1/displaylink-installer.sh: No such file or directory

This is the file:
> stat 4.1/displaylink-driver-4.1.9.run 
  File: 4.1/displaylink-driver-4.1.9.run
  Size: 14274265        Blocks: 27880      IO Block: 4096   regular file
Device: 812h/2066d      Inode: 20973059    Links: 1
Access: (0700/-rwx------)  Uid: (    0/    root)   Gid: (    0/    root)

redjamesg, 2018-01-02:
See if this fixes the issue.
https://github.com/AdnanHodzic/displaylink-debian/pull/108/commits

AdnanHodzic, 2018-01-03:
@debalance merge request made by @redjamesg has just been merged into master. Please re-open this issue if this hasn't solved your problem.

closed 2018-01-03 by AdnanHodzic

debalance, 2018-01-03:
All good now, thx!