AdnanHodzic/displaylink-debian #110: listprovider empty

plombix, 2018-01-16:
I have a debian : 
`Linux debian 4.9.0-5-amd64 #1 SMP Debian 4.9.65-3+deb9u2 (2018-01-04) x86_64 GNU/Linux`

service is running : 
`● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-01-16 16:06:17 CET; 4min 46s ago
  Process: 655 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.1.9 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 663 (DisplayLinkMana)
    Tasks: 32 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─663 /opt/displaylink/DisplayLinkManager
`

All instalation went **great** ( it was a sign,  i should know that now -_-) .

BUT (there it is)  the damm thing is not detected  : 

`
•100% ➜ xrandr --listproviders                                         
Providers: number : 0

~ 
•100% ➜ xrandr                
Screen 0: minimum 320 x 200, current 1366 x 768, maximum 8192 x 8192
XWAYLAND0 connected 1366x768+0+0 (normal left inverted right x axis y axis) 280mm x 160mm
   1366x768      59.98*+
`
Dmesg however  shows  it : 

`[ +10.414559] usb 4-2: new SuperSpeed USB device number 3 using xhci_hcd
[  +0.020743] usb 4-2: New USB device found, idVendor=17e9, idProduct=ff03
[  +0.000003] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 4-2: Product: MB168B
[  +0.000002] usb 4-2: Manufacturer: DisplayLink
[  +0.000001] usb 4-2: SerialNumber: G4LMTF160122
[  +6.620679] evdi: [D] evdi_painter_connect:572 (dev=1) Connected with ffffa09a2f231c00
[  +0.000002] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[  +0.000079] evdi: [D] evdi_painter_crtc_state_notify:481 (dev=1) Notifying crtc state: 3`

What am i missing & /  or  doing wrong   ?

redjamesg, 2018-01-16:
Did you try the issues addressed in the post-install?
I think issue #68 might be what is causing your problems.

gregchi, 2018-01-16:
It does not work with wayland. Do a test and try for example GNOME Classic or any other desktop enviroment working on Xorg server - not Wayland.

plombix, 2018-01-17:
After xorg messing time , still not detected , howether switch of  wayland makes providerlist to  1.

plombix, 2018-01-17:
So i finaly got it work .
### enable xorg  (after getting off from wayaland)

switch to tty6 (it's possible @ login screen)

> /!\  /!\ switch to tty6 == 'ctrl + shift + f6'  /!\  /!\ 

got in xorg folder   (as root)  `cd /etc/X11/ ` 
init autoconfig  `Xorg -configure` ou `X -configure`
copy the 'xorg.conf.new' freshly created from my home  to xorg folder (current folder at this point) as 
''xorg.conf'
then reboot .
#### black screen? 
xorg has crash  : there is an othrer problem
switch to tty6 (ctrl + shift + f6)  and delete file 'xorg.conf'
and then, reboot _again_ 

### after normal login ( gnome not wayland )
 `xrandr --listproviders` got  me a better output 
```
Providers: number : 2
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel
Provider 1: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```
then i could get back with normal procedure of [post install ](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

thanks @gregchi  and @redjamesg  for your time, you got me in good direction :+1:  :+1:

closed 2018-01-17 by plombix