AdnanHodzic/displaylink-debian #111: displaylink-debian.sh not running on fresh install of linux mint 18.3 Cinnamon 64 bit

musabmian, 2018-01-29:
displaylink-debian.sh not running on fresh install of linux mint 18.3 Cinnamon 64 bit
Cinnamon Version 3.6.6
Linux Kernel 4.10.0-38 generic

when i run the script sudo ./displaylink-debian.sh  I get
./displaylink-debian.sh: 8: ./displaylink-debian.sh: Syntax error: newline unexpected

can anyone help with this I am fairly new to linx

redjamesg, 2018-01-29:
Look at #89 and

musabmian, 2018-01-29:
sudo displaylink-debian.sh returns command not found

musabmian, 2018-01-29:
ref to #65 
git reset --hard && git pull returns the program git is currently not installed so I installed it

again ran sudo ./displaylink-debian.sh and got this
/displaylink-debian.sh: 8: ./displaylink-debian.sh: Syntax error: newline unexpected

musabmian, 2018-01-29:
will try again after installing update files

redjamesg, 2018-01-29:
What if you change to root and run the script, without sudo interpreter.
# su
# ./displaylink-debian.sh

musabmian, 2018-01-29:
su returns this

musab-2 DL # ./displaylink-debian.sh
./displaylink-debian.sh: line 7: syntax error near unexpected token `newline'
./displaylink-debian.sh: line 7: `<!DOCTYPE html>'
musab-2 DL #

musabmian, 2018-01-29:
please help I am stuck

redjamesg, 2018-01-29:
What does it display with
$ which bash
And maybe see what it tells you with
$ file `/bin/bash`

musabmian, 2018-01-29:
musab@musab-2 ~ $ which bash
/bin/bash
musab@musab-2 ~ $ file /bin/bash
/bin/bash: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=01f82e2c5ecef569ec9e29a8c392d9f10d2eaeaf, stripped

redjamesg, 2018-01-29:
What does
$ ls -l displaylink-debian.sh
Show, I am wondering if the executable bit is set on the file.

musabmian, 2018-01-29:
yes i went to properties/permissions and set it to allow execute as program earlier

musabmian, 2018-01-29:
musab@musab-2 ~/Downloads/DL $ ls -l displaylink-debian.sh
-rwxr-xr-x 1 musab musab 141922 Jan 29 12:38 displaylink-debian.sh

musabmian, 2018-01-29:
should i update to this..?
Packages included in this update: linux-headers-4.13.0-32 linux-headers-4.13.0-32-generic linux-image-4.13.0-32-generic linux-image-extra-4.13.0-32-generic

redjamesg, 2018-01-29:
Yes those should be updated.

redjamesg, 2018-01-29:
Every package claiming to be upgradeable better be upgraded.

musabmian, 2018-01-29:
[image]

musabmian, 2018-01-29:
[image]

musabmian, 2018-01-29:
Updated all available
restarted
gave command again but still same result
musab@musab-2 ~/Downloads/DL $ sudo ./displaylink-debian.sh
[sudo] password for musab: 
./displaylink-debian.sh: 8: ./displaylink-debian.sh: Syntax error: newline unexpected

tried with SU
musab-2 DL # ./displaylink-debian.sh
./displaylink-debian.sh: line 7: syntax error near unexpected token `newline'
./displaylink-debian.sh: line 7: `< !DOCTYPE html >'

musabmian, 2018-01-29:
sorry closed by mistake

adangel, 2018-01-29:
It looks like that the script `displaylink-debian.sh` is corrupt. Can you execute `md5sum displaylink-debian.sh` and report the output here.

Example execution:

```
$ md5sum displaylink-debian.sh 
0f7b5f4b85806d685c703f7d26635ca0  displaylink-debian.sh
```

If the md5sum doesn't match, then your copy of the script is broken. How did you download it?

musabmian, 2018-01-29:
musab@musab-2 ~/Downloads/DL $ md5sum displaylink-debian.sh
b1a382a6fe0b076245e0b58ba6662a49  displaylink-debian.sh

musabmian, 2018-01-29:
i downloaded it from this page
will try to dl again and try
thank you

musabmian, 2018-01-29:
deleted the old one
downloaded a new copy

same result
musab@musab-2 ~/Downloads $ sudo ./displaylink-debian.sh
./displaylink-debian.sh: 8: ./displaylink-debian.sh: Syntax error: newline unexpected
musab@musab-2 ~/Downloads $ md5sum displaylink-debian.sh
3c1089df083fa43fc9264e7f2190328c  displaylink-debian.sh

adangel, 2018-01-29:
You can also download it as a ZIP file from here: https://github.com/AdnanHodzic/displaylink-debian/archive/master.zip

After you have downloaded it, you'll need to unzip it, e.g.:

```
$ unzip displaylink-debian-master.zip 
Archive:  displaylink-debian-master.zip
075594536fe4683a5e25aec99e3b6379662ef2ea
   creating: displaylink-debian-master/
  inflating: displaylink-debian-master/README.md  
  inflating: displaylink-debian-master/displaylink-debian.sh  
  inflating: displaylink-debian-master/post-install-guide.md  
$ md5sum displaylink-debian-master/displaylink-debian.sh 
0f7b5f4b85806d685c703f7d26635ca0  displaylink-debian-master/displaylink-debian.sh

```

Note: the script will be then in the subfolder `displaylink-debian-master`.

adangel, 2018-01-29:
It's probably your browser, that messes things up. Are you using this page: https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh for downloading (menu file, save as)? Otherwise you'll save a HTML page and not a script...

musabmian, 2018-01-29:
Thank you so much

the zip worked will keep you updated

musab@musab-2 ~/Downloads $ sudo ./displaylink-debian.sh

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux
* Deepin

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

musabmian, 2018-01-29:
Checking dependencies

unzip is installed
linux-headers-4.13.0-32-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: package 'linux-source' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-source not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-source-4.4.0
Suggested packages:
  libncurses-dev | ncurses-dev kernel-package libqt3-dev
Recommended packages:
  libc-dev
The following NEW packages will be installed:
  linux-source linux-source-4.4.0
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 112 MB of archives.
After this operation, 130 MB of additional disk space will be used.
Do you want to continue? [Y/n] y

musabmian, 2018-01-29:
Installing driver version: 4.1

DisplayLink Linux Software 4.1.9 install script called: install
Distribution discovered: Linux Mint 18.3 Sylvia
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing licence file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

-------------------------------------------------------------------

Performing post install steps

Created symlink from /etc/systemd/system/multi-user.target.wants/dlm.service to /lib/systemd/system/dlm.service.

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_4.1.zip" file
Removing redundant: "4.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

musabmian, 2018-01-29:
this works like a charm 
Thank you Adnan Hodzic  redjamesg and adangel :)

musabmian, 2018-01-29:
so it installed correctly and i restarted the system 
then went to display and chose my other screen turned it on and it worked 
then I made it primary and removed my temp screen that i was using during install  and rebooted
the displaylink screen did not turn on
so i reattached the temp screen and rebooted and went to check is the service was running systemctl status dlm.service  it was running
now when i try to turn on the other screen this is what i get
[image]

PR #112 (2018-01-29): Add troubleshooting section for "syntax error near unexpected token
Add troubleshooting section for "syntax error near unexpected token \`newline'"

Fixes #111, Fixes #102
Refs #89, Refs #65
commit 24f26b55 (2018-01-29): Add troubleshooting section for "syntax error near unexpected token \`newline'"

Fixes #111, Fixes #102
Refs #89, Refs #65
closed 2018-01-29 by musabmian