AdnanHodzic/displaylink-debian #113: Getting error when connecting to new screen

musabmian, 2018-01-29:
Mint 18.3 Cinnamon 64 bit
Cinnamon version 3.6.7
linux kernel 4.10.0-38 generic

so i was able to run the script and install the drivers 
i followed the post install procedure and checked to make sure displaylink was running and it was
then i connected to the new screen and the first time it works but when i try to make it primary and apply it freezes and after that it never connects and i get this message 

[image]

musabmian, 2018-01-30:
can someone please point out if i missed something in the process..?
I tried it from scratch again today and did a reboot after installation
checked the service and it was running
went to display and turned on the usb monitor it worked
i made it primary and it worked
then i unplugged my temp monitor and the usb monitor tuned off for a sec and turned back on
i then restarted the system and the usb monitor never turned on just the power light turned on and off a couple of times
then i turned it off again and plugged my temp monitor back and restarted and check the service it wasnt running. so enabled it and ran it.
when i tried to use the usb monitor again it gave me the same error as above.
Thanks in advance for any help

adangel, 2018-01-30:
Disclaimer: I don't use it anymore, since wayland support is missing.

When I used it, I made the experience, that I maybe needed to manually unplug and plug the usb displaylink device, before it started to work again after reboot/standby. Also note, that it takes a while after you have plugged in the displaylink device, before it recognizes the monitor.
Also see, whether the [Post Install Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) is of any help.

musabmian, 2018-01-30:
the service does not start after reboot   systemctl status dlm.service
i have enabled it as described in the post guide  systemctl enable dlm.service
but after restart it still isnt running
an work arround to this..?

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic