AdnanHodzic/displaylink-debian #114: Not installing with new kernel (linux-headers-4.15.0-1-amd64)

aowrew-fp, 2018-02-22:
Dear Maintainer, 
there seems to be a problem with running/installing the service on Debian SID after latest kernel update. Please take a look at the logs:

> Installing driver version: 4.1
> DisplayLink Linux Software 4.1.9 install script called: install
> Distribution discovered: Debian GNU/Linux unstable (sid)
> Installing
> Configuring EVDI DKMS module
> Registering EVDI kernel module with DKMS
> Building EVDI kernel module with DKMS
> ERROR (code 3): Failed to build evdi/4.1.9. Consult /var/lib/dkms/evdi/4.1.9/build/make.log for details..

The make.log content:

```shell
DKMS make.log for evdi-4.1.9 for kernel 4.15.0-1-amd64 (x86_64)
Thu 22 Feb 09:18:54 CET 2018
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/4.1.9/build SRCROOT=/var/lib/dkms/evdi/4.1.9/build CONFIG_MODULE_SIG= -C /lib/modules/4.15.0-1-amd64/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-1-amd64'
make -C /usr/src/linux-headers-4.15.0-1-amd64 KBUILD_SRC=/usr/src/linux-headers-4.15.0-1-common \
-f /usr/src/linux-headers-4.15.0-1-common/Makefile modules
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/lib/dkms/evdi/4.1.9/build/.tmp_versions ; rm -f /var/lib/dkms/evdi/4.1.9/build/.tmp_versions/*
make -f /usr/src/linux-headers-4.15.0-1-common/scripts/Makefile.build obj=/var/lib/dkms/evdi/4.1.9/build
   gcc-7 -Wp,-MD,/var/lib/dkms/evdi/4.1.9/build/.evdi_drv.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include -I./arch/x86/include/generated  -I/usr/src/linux-headers-4.15.0-1-common/include -I./include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-4.15.0-1-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-4.15.0-1-common/include/linux/kconfig.h  -I/var/lib/dkms/evdi/4.1.9/build -I/var/lib/dkms/evdi/4.1.9/build -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-var-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init  -I/usr/src/linux-headers-4.15.0-1-common/include/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_drv"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/4.1.9/build/.tmp_evdi_drv.o /var/lib/dkms/evdi/4.1.9/build/evdi_drv.c
   gcc-7 -Wp,-MD,/var/lib/dkms/evdi/4.1.9/build/.evdi_modeset.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include -I./arch/x86/include/generated  -I/usr/src/linux-headers-4.15.0-1-common/include -I./include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-4.15.0-1-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-4.15.0-1-common/include/linux/kconfig.h  -I/var/lib/dkms/evdi/4.1.9/build -I/var/lib/dkms/evdi/4.1.9/build -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-var-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init  -I/usr/src/linux-headers-4.15.0-1-common/include/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_modeset"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/4.1.9/build/.tmp_evdi_modeset.o /var/lib/dkms/evdi/4.1.9/build/evdi_modeset.c
   gcc-7 -Wp,-MD,/var/lib/dkms/evdi/4.1.9/build/.evdi_connector.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include -I./arch/x86/include/generated  -I/usr/src/linux-headers-4.15.0-1-common/include -I./include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-4.15.0-1-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-4.15.0-1-common/include/linux/kconfig.h  -I/var/lib/dkms/evdi/4.1.9/build -I/var/lib/dkms/evdi/4.1.9/build -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-var-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init  -I/usr/src/linux-headers-4.15.0-1-common/include/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_connector"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/4.1.9/build/.tmp_evdi_connector.o /var/lib/dkms/evdi/4.1.9/build/evdi_connector.c
   gcc-7 -Wp,-MD,/var/lib/dkms/evdi/4.1.9/build/.evdi_encoder.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/src/linux-headers-4.15.0-1-common/arch/x86/include -I./arch/x86/include/generated  -I/usr/src/linux-headers-4.15.0-1-common/include -I./incl [... cut ...]
  if [ "-pg" = "-pg" ]; then if [ /var/lib/dkms/evdi/4.1.9/build/evdi_gem.o != "scripts/mod/empty.o" ]; then ./scripts/recordmcount  "/var/lib/dkms/evdi/4.1.9/build/evdi_gem.o"; fi; fi;
make[3]: *** [/usr/src/linux-headers-4.15.0-1-common/Makefile:1523: _module_/var/lib/dkms/evdi/4.1.9/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-1-amd64'
make: *** [Makefile:22: all] Error 2
```

Switching back to 4.14.* works just fine. 

It this something that can be fixed as part of your code or do we have to wait for a driver update?

Best,
Alex

redjamesg, 2018-02-22:
The error has something to do with the kernel source and is not related to EVDI source, the proof is in the build error:

> echo >&2 "  ERROR: Kernel configuration is invalid.";		\
> echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
> echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";

aowrew-fp, 2018-02-23:
Hello @redjamesg - turn out it actually was an EVDI source issue, please see the following link: https://github.com/DisplayLink/evdi/issues/117 - there's a typo in the EVDI source. Changing the 16 to 15 worked a charm and I now have both my additional screens working fine.

Thanks!

closed 2018-02-23 by aowrew-fp