AdnanHodzic/displaylink-debian #116: Latest Debian on NUC - install ok, no workee

MonteShaffer, 2018-02-26:
Hi,

First, thanks for taking the effort to make this work.  I reviewed your code and must say well done.

I am developing an exercise platform for patients with Parkinson's Disease, and am hoping to use multiple USB touchscreen monitors, so displaylink-debian is very important for my project.

The system `sudo dmidecode -t system`

```
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Intel Corporation
	Product Name: NUC6CAYH
	Version: J26843-405
	Serial Number: G6AY80200DGQ
	UUID: DA6ACC15-2571-5721-56AF-94C69119B46F
	Wake-up Type: Power Switch
	SKU Number:                                  
	Family: Intel NUC

Handle 0x0023, DMI type 12, 5 bytes
System Configuration Options
	Option 1: Default string

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

This NUC has an onboard Intel HD graphics card.  It also has stable Linux boot options **within UEFI mode** which was a big reason I selected this system; the display `sudo lshw -class display`

```
  *-display                 
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 0b
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:376 memory:90000000-90ffffff memory:80000000-8fffffff ioport:f000(size=64) memory:c0000-dffff
```

I installed `debian-live-9.3.0-amd64-mate+nonfree.iso` which gave me no driver issues (even for the wifi/bluetooth) for the Intel Install.  So in my scenario, I am using **lightdm** and **mate**.

So I got my Xenarc (DisplayLink-compliant) touchscreen today (https://www.xenarcdirect.com/708TSU.html)

I have attached the USB touchscreen via a powered 2.0 HUB (https://www.amazon.com/gp/product/B00HL7Z46K/) and it appears to be listed `sudo lsusb`

 ```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8087:0aa7 Intel Corp. 
Bus 001 Device 009: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 008: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 005: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 003: ID 0bc2:231a Seagate RSS LLC 
Bus 001 Device 006: ID 0eef:0001 D-WAV Scientific Co., Ltd eGalax TouchScreen
Bus 001 Device 004: ID 17e9:0199 DisplayLink 
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I ran your script, worked as expected.

I ran the "post-install" stuff, and this is where I am stuck.

So it appeared to install, and after reboot, I proceed with the post-installation, `systemctl status dlm.service` shows "green active"

```
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enab
   Active: active (running) since Mon 2018-02-26 16:37:00 EST; 32min ago
  Process: 667 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.1.9
 Main PID: 672 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─672 /opt/displaylink/DisplayLinkManager

Feb 26 16:37:00 ats systemd[1]: Starting DisplayLink Manager Service...
Feb 26 16:37:00 ats systemd[1]: Started DisplayLink Manager Service.
```

When I run `xrandr --listproviders` I only get one, the main display.  I installed `arandr` to manage from a GUI, but we are not getting any info.

```
xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting
```

So I tried the suggestions in Issue #99 to unload evdi

```
systemctl stop dlm.service
rmmod -f evdi
systemctl start dlm.service
```

again without luck, still not working...

When I review `dmesg` I am getting a lot of information about interface claimed by:

```
[ 2464.347534] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2465.354328] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2466.360883] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2467.367762] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2468.374341] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2469.381184] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2470.387583] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2471.394523] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2472.401198] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2473.407756] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[ 2474.414538] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
```

From this page (https://forum.manjaro.org/t/solved-displaylink-support-on-manjaro/10475/7), I saw that it may be an issue with other USB devices I have connected, and with "usb-storage" it may be my external hard drive which **is not** plugged into the USB 2.0 powered port, but is plugged into a USB 3.0 port directly on the NUC.

I unplugged the external hard drive, rebooted, and tried again.  Still no workee and `dmesg` is unchanged.

```
[  176.072436] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[  177.078734] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[  178.085708] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[  179.092519] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[  180.100156] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
```

So I am thinking this is an issue with the core driver and "udev".

I installed `sudo apt-get install libusb-dev` but it appears there is no direct command-line access to libusb protocols, I have to code something in `C` which is leaving my domain of expertise.

Do you think `dmesg` is indicating the nature of the error?  If so, how to fix?  Or maybe it is about mate and lightdm?

Thoughts?  

monte

{x:

redjamesg, 2018-02-26:
What if you take the powered USB ports out of the question, so it would be 
NUC 》 Xenarc
Just to eliminate any bad combo of a USB-3 》USB-2 》USB-3 conversion that might create problems.

At the same time I must add the reference of mate playing well with the displaylink driver is sparse.

MonteShaffer, 2018-02-27:
After doing a bit more digging... `fdisk`

```
Device         Start       End   Sectors   Size Type
/dev/sda1       2048   1050623   1048576   512M EFI System
```

Note the filesize... and `lsusb -t`

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 5: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 5: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 6: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 8: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
```

I believe the conflict is with the /boot/efi (for UEFI reasons) and the DisplayLink, something similar to an IRQ conflict.

```
  |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```
I believe 480M, usb-storage is the 512M EFI system.  It is reserved on that IRQ and DisplayLink is trying to take it over.  Or the Touchscreen is... 

```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0aa7 Intel Corp. 
Bus 001 Device 008: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 007: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 011: ID 06c2:003f Phidgets Inc. (formerly GLAB) 
Bus 001 Device 003: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 006: ID 0eef:0001 D-WAV Scientific Co., Ltd eGalax TouchScreen
Bus 001 Device 004: ID 17e9:0199 DisplayLink 
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

which still shows in `dmesg`

```
[15855.405494] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[15856.412012] usb 1-1.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
```

MonteShaffer, 2018-02-27:
@redjamesg Thanks for the info, the updates are indeed with the Xenarc being directly attached to a USB port on the NUC without any external hub.

MonteShaffer, 2018-02-27:
@redjamesg If "mate" on debian is not a great solution, what is a better alternative?  I am looking for basic GUI support not "bleeding edge" that unnecessarily uses valuable resources.  

I want to use Debian and support DisplayLink.  

GNOME?  KDE?  Xfce?  LXDE? e17 Enlightenment (X11->Wayland)?

I am in development, so I can rebuild the machine, trying each, but any suggestions would be invaluable.

AdnanHodzic, 2018-02-27:
@MonteShaffer UEFI definitely has to be disabled, reference: [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues)

And as of this point I would stick with X11 as upstream hasn't added support for Wayland as of yet.

Rest of it shouldn't matter that much.

MonteShaffer, 2018-02-27:
@redjamesg About UEFI, I don't know that is a killer, since it is tied to `dlm.service` starting.  I am getting my service to start fine.  Intel has been developing a UEFI that is Linux compliant for some time.  I am not convinced yet UEFI is the culprit.

My USB-powered hub technically consists of two independent 7-port hubs.  So I plugged Xenarc into a separate hub, and `et oila` I was able to get `xrandr --listproviders` to list two devices (although I did not see any option in `arandr` for a second monitor.  So I am making progress.

In the end, I need multiple touchscreens playing nice on all hubs, so I need to dig into this error more.

So I setup `Xorg -config` and `udevadm monitor` to see what is going on.  The conflict is with "usb-storage" on the hub controller.  So I grep'd the string in single user mode (no GUI) using  `grep -Irnw '/path' -e 'usb-storage' > /mnt/data/_play_/grep-path.txt` and stored on a different device ... I reviewed several paths which is summarized:

```
/root/.viminfo:32:?/usb-storage
/root/.viminfo:33:|2,1,1519732014,47,"usb-storage"
/run/udev/data/b11:1:23:E:ID_USB_DRIVER=usb-storage
/run/udev/data/b11:0:25:E:ID_USB_DRIVER=usb-storage
/etc/usb_modeswitch.conf:26:# Optional increase of "delay_use" for the usb-storage driver; there are hints
/sys/kernel/debug/usb/devices:37:I:* If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
/sys/kernel/debug/usb/devices:99:I:* If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
/sys/devices/pci0000:00/0000:00:15.0/usb1/1-4/1-4.7/1-4.7.3/1-4.7.3.2/1-4.7.3.2:2.0/uevent:2:DRIVER=usb-storage
/sys/devices/pci0000:00/0000:00:15.0/usb1/1-4/1-4.7/1-4.7.3/1-4.7.3.2/1-4.7.3.2:2.0/host4/scsi_host/host4/proc_name:1:usb-storage
/sys/devices/pci0000:00/0000:00:15.0/usb1/1-4/1-4.2/1-4.2.2/1-4.2.2:2.0/uevent:2:DRIVER=usb-storage
/sys/devices/pci0000:00/0000:00:15.0/usb1/1-4/1-4.2/1-4.2.2/1-4.2.2:2.0/host2/scsi_host/host2/proc_name:1:usb-storage
/proc/kmsg:1:<4>[  130.171644] usb 1-4.2.2: usbfs: interface 0 claimed by usb-storage while 'DisplayLinkMana' sets config #1
/proc/kmsg:2:<4>[  130.174234] usb 1-4.7.3.2: usbfs: interface 0 claimed by usb-storage while 'DisplayLinkMana' sets config #1
/proc/sched_debug:166:     usb-storage   374     10114.026494      1827   120         0.000000        39.901864         0.000000 0 0 /
/proc/sched_debug:259:     usb-storage   633      8220.223225      1820   120         0.000000        44.587178         0.000000 0 0 /
/proc/374/task/374/status:1:Name:	usb-storage
/proc/374/task/374/sched:1:usb-storage (374, #threads: 1)
/proc/374/task/374/comm:1:usb-storage
/proc/374/task/374/stat:1:374 (usb-storage) S 2 0 0 0 -1 2129984 0 0 0 0 3 0 0 0 20 0 1 0 325 0 0 18446744073709551615 0 0 0 0 0 0 0 2147483647 0 0 0 0 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0
/proc/374/status:1:Name:	usb-storage
/proc/374/sched:1:usb-storage (374, #threads: 1)
/proc/374/comm:1:usb-storage
/proc/374/stat:1:374 (usb-storage) S 2 0 0 0 -1 2129984 0 0 0 0 3 0 0 0 20 0 1 0 325 0 0 18446744073709551615 0 0 0 0 0 0 0 2147483647 0 0 0 0 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0
/proc/633/task/633/status:1:Name:	usb-storage
/proc/633/task/633/sched:1:usb-storage (633, #threads: 1)
/proc/633/task/633/comm:1:usb-storage
/proc/633/task/633/stat:1:633 (usb-storage) S 2 0 0 0 -1 2129984 0 0 0 0 4 0 0 0 20 0 1 0 402 0 0 18446744073709551615 0 0 0 0 0 0 0 2147483647 0 0 0 0 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0
/proc/633/status:1:Name:	usb-storage
/proc/633/sched:1:usb-storage (633, #threads: 1)
/proc/633/comm:1:usb-storage
/proc/633/stat:1:633 (usb-storage) S 2 0 0 0 -1 2129984 0 0 0 0 4 0 0 0 20 0 1 0 402 0 0 18446744073709551615 0 0 0 0 0 0 0 2147483647 0 0 0 0 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0

/lib/modules/4.9.0-4-amd64/modules.order:918:kernel/drivers/usb/storage/usb-storage.ko
/lib/modules/4.9.0-4-amd64/modules.dep:917:kernel/drivers/usb/storage/uas.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:918:kernel/drivers/usb/storage/usb-storage.ko: kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:919:kernel/drivers/usb/storage/ums-alauda.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:920:kernel/drivers/usb/storage/ums-cypress.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:921:kernel/drivers/usb/storage/ums-datafab.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:922:kernel/drivers/usb/storage/ums-eneub6250.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
/lib/modules/4.9.0-4-amd64/modules.dep:923:kernel/drivers/usb/storage/ums-freecom.ko: kernel/drivers/usb/storage/usb-storage.ko kernel/drivers/usb/core/usbcore.ko kernel/drivers/usb/common/usb-common.ko kernel/drivers/scsi/scsi_mod.ko
[... cut ...]
        |__ Port 6: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 6: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 8: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M

```

MonteShaffer, 2018-02-27:
To clarify, when I had the Xenarc on its own hub, 

```
systemctl status dlm.service
xrandr --listproviders
xrandr --setprovideroutputsource 1 0
```

worked and 

```
xrandr --setprovideroutputsource 2 0
```

did not work.

MonteShaffer, 2018-02-27:
Xorg is no longer installed by default any more, so

```
/sbin/init 1

Xorg -configure

cp /root/xorg.conf.new /etc/X11/xorg.conf

reboot
```

I am trying to figure out `udev` in relationship to `displaylink` and `evdi` and `libdlo` (which is `libusb-dev`)

I extracted the components of the driver `./displaylink-driver-4.1.9.run --noexec --keep` based on this information (https://support.displaylink.com/knowledgebase/articles/679060)

Some import elements of the partial src:

```
COREDIR=/opt/displaylink
LOGSDIR=/var/log/displaylink
PRODUCT="DisplayLink Linux Software"
VERSION=4.1.9

add_udev_rule()
{
  echo "Adding udev rule for DisplayLink DL-3xxx/5xxx devices"
  create_udev_rules_file /etc/udev/rules.d/99-displaylink.rules
  udevadm control -R
  udevadm trigger
}

echo "Registering EVDI kernel module with DKMS"
  dkms add evdi/$MODVER -q
  if [ $? != 0 -a $? != 3 ]; then
    echo "Unable to add evdi/$MODVER to DKMS source tree." > $ERRORS
    return 2
  fi

. /opt/displaylink/udev.sh

exec /opt/displaylink/DisplayLinkManager

###

cat /sys/bus/usb/devices/*/idVendor | grep 17e9 | wc -l

for f in $(find $root/displaylink/by-id -type l -exec realpath {} \; 2> /dev/$
    test -c $f && echo $f;
  done | wc -l

```

MonteShaffer, 2018-02-27:
Any thoughts on the DisplayLink log (is it a simple base64 encode)?

`less /var/log/displaylink/DisplayLinkManager.log`

MonteShaffer, 2018-03-06:
So I installed without UEFI (installed with legacy BIOS), getting same error:

```
usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
```

Below is the entire boot dmesg:

```
dmesg
[    0.000000] Linux version 4.9.0-4-amd64 (debian-kernel@lists.debian.org) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 SMP Debian 4.9.65-3 (2017-12-03)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.9.0-4-amd64 root=/dev/sda1 ro quiet
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009b3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009b400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x000000007736ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000077370000-0x0000000079491fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079492000-0x00000000794acfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000794ad000-0x00000000798b0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000798b1000-0x0000000079c3ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079c40000-0x0000000079fc2fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079fc3000-0x0000000079fc3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079fc4000-0x0000000079fedfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079fee000-0x000000007a9defff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a9df000-0x000000007a9e0fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a9e1000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0 present.
[    0.000000] DMI: Intel Corporation NUC6CAYH/NUC6CAYB, BIOS AYAPLCEL.86A.0043.2017.1123.1559 11/23/2017
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-6FFFF write-back
[    0.000000]   70000-7FFFF uncachable
[    0.000000]   80000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7F80000000 write-back
[    0.000000]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000000]   2 base 007B000000 mask 7FFF000000 uncachable
[    0.000000]   3 base 0100000000 mask 7F00000000 write-back
[    0.000000]   4 base 0200000000 mask 7F80000000 write-back
[    0.000000]   5 base 00FF000000 mask 7FFF000000 write-combining
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[... 787 lines cut ...]
[   14.699960] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   15.040182] random: crng init done
[   15.704162] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   16.707399] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   17.711319] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   18.715087] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   19.720339] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   20.728182] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   21.735032] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   22.741420] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[... cut ...]
[   32.800604] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   33.807318] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   34.813153] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   35.820177] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
[   36.825746] usb 1-4.7.2.2: usbfs: interface 0 claimed by usb-storage while 'DeviceManager_0' sets config #1
```

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Could you please run re-install with latest changes and report if problem persists?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic