AdnanHodzic/displaylink-debian #119: Displaylink service won't start

kfleten, 2018-03-12:
Hi
I have a Ubuntu 16.04 with UEFI where displaylink won't start. I know UEFI boot is an issue, but I belive it should work if Secure Boot is disabled using mokutil, but the result is the same. The system is a Dell XPS 13 with Intel i915 video driver. I have read that Intel drivers might cause issues. 
Is there anyone that knows what to do ?

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic