AdnanHodzic/displaylink-debian #120: Debian (stretch) login loop

Kaffiend, 2018-03-14:
Installed fine. No errors but now im in a login loop.

RichardSouta, 2018-03-20:
Happend to me as well on Ubuntu 18.04.

AdnanHodzic, 2018-03-20:
@Kaffiend please confirm you're using Wayland.

@RichardSouta are you using Xorg (default) or Wayland?

Could both include displaylink-debian install output as well as the GDM logs?

Kaffiend, 2018-03-20:
I’ve tried both xorg and wayland, with and without proprietary NVidia drivers on a 1060ti on each attempt. When I get home I’ll run through it again and grab more logging.

Ian Mackie
Lead Developer
Complete System Support, Inc.

From: Adnan Hodzic
Sent: Tuesday, March 20, 2018 1:12 PM
To: AdnanHodzic/displaylink-debian
Cc: Ian Mackie; Mention
Subject: Re: [AdnanHodzic/displaylink-debian] Debian (stretch) login loop(#120)

@Kaffiend please confirm you're using Wayland.
@RichardSouta are you using Xorg (default) or Wayland?
Could both include displaylink-debian install output as well as the GDM logs?

RichardSouta, 2018-03-22:
@AdnanHodzic I'm using Xorg.
`--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux
* Deepin

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.15.0-12-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: balík „linux-source“ není nainstalován a není dostupná žádná informace
Použijte dpkg --info (= dpkg-deb --info) k prozkoumání archivu
a dpkg --contents (= dpkg-deb --contents) k vypsání jeho obsahu.
linux-source not found! Install? [Y/n] y
Načítají se seznamy balíků… Hotovo
Vytváří se strom závislostí       
Načítají se stavové informace… Hotovo
Následující dodatečné balíky budou instalovány:
  linux-source-4.15.0
Navrhované balíky:
  libncurses-dev | ncurses-dev kernel-package libqt3-dev
Následující NOVÉ balíky budou nainstalovány:
  linux-source linux-source-4.15.0
0 aktualizováno, 2 nově instalováno, 0 k odstranění a 0 neaktualizováno.
Nutno stáhnout 128 MB archivů.
Po této operaci bude na disku použito dalších 144 MB.
Chcete pokračovat? [Y/n] y
Stahuje se:1 http://cz.archive.ubuntu.com/ubuntu bionic/main amd64 linux-source-4.15.0 all 4.15.0-12.13 [128 MB]
Stahuje se:2 http://cz.archive.ubuntu.com/ubuntu bionic/main amd64 linux-source all 4.15.0.12.13 [2 276 B]
Staženo 128 MB za 2min 47s (769 kB/s)                                          
Vybírá se dosud nevybraný balík linux-source-4.15.0.
(Načítá se databáze … nyní je nainstalováno 137397 souborů a adresářů.)
Připravuje se nahrazení …/linux-source-4.15.0_4.15.0-12.13_all.deb …
Rozbaluje se linux-source-4.15.0 (4.15.0-12.13) …
Vybírá se dosud nevybraný balík linux-source.
Připravuje se nahrazení …/linux-source_4.15.0.12.13_all.deb …
Rozbaluje se linux-source (4.15.0.12.13) …
Nastavuje se balík linux-source-4.15.0 (4.15.0-12.13) …
Nastavuje se balík linux-source (4.15.0.12.13) …

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement
available at http://www.displaylink.com/downloads/file?id=1123
and accept here: [Y]es or [N]o: y

Downloading DisplayLink Ubuntu driver:

--2018-03-22 17:02:36--  http://www.displaylink.com/downloads/file?id=1123
Překládám www.displaylink.com (www.displaylink.com)… 52.208.143.240, 34.252.223.12, 54.76.51.192, ...
Navazuje se spojení s www.displaylink.com (www.displaylink.com)|52.208.143.240|:80… spojeno.
HTTP požadavek odeslán, program čeká na odpověď… 303 See Other
Přesměrováno na: http://assets.displaylink.com/live/downloads/software/f1123_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1521735173&Signature=xNchSC0qYOyo6r8xAnQe01Iboho%3D [následuji]
--2018-03-22 17:02:37--  http://assets.displaylink.com/live/downloads/software/f1123_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%204.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1521735173&Signature=xNchSC0qYOyo6r8xAnQe01Iboho%3D
Překládám assets.displaylink.com (assets.displaylink.com)… 52.218.64.52
Navazuje se spojení s assets.displaylink.com (assets.displaylink.com)|52.218.64.52|:80… spojeno.
HTTP požadavek odeslán, program čeká na odpověď… 200 OK
Délka: 14112877 (13M) [application/octet-stream]
Ukládám do: „DisplayLink_Ubuntu_4.2.zip“

DisplayLink_Ubuntu_ 100%[===================>]  13,46M   640KB/s    za 24s     

2018-03-22 17:03:01 (575 KB/s) – „DisplayLink_Ubuntu_4.2.zip“ uloženo [14112877/14112877]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_4.2.zip
  inflating: 4.2/displaylink-driver-4.2.29.run  
  inflating: 4.2/LICENSE             
   creating: 4.2/__MACOSX/
  inflating: 4.2/__MACOSX/._LICENSE  
Creating directory displaylink-driver-4.2.29
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 4.2.29  100%  

## NO suitable EVDI source found ##

-------------------------------------------------------------------

Installing driver version: 4.2

DisplayLink Linux Software 4.2.29 install script called: install
Distribution discovered: Ubuntu Bionic Beaver (development branch)
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing licence file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

-------------------------------------------------------------------

Performing post install steps

Created symlink /etc/systemd/system/multi-user.target.wants/dlm.service → /lib/systemd/system/dlm.service.

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_4.2.zip" file
Removing redundant: "4.2" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------
`

redjamesg, 2018-03-22:
That error is due to a quick fix I introduced to correct a miss typed variable on EVDI source version 4.1.19, since the latest is 4.2.29 and not matching 4.1.19 this error is displayed.

It has no effect on the build of the module what so ever, which is why the script continues and installs the EVDI module.

sininspira2, 2018-03-22:
I'm having this same issue with Kali Rolling (2018.1). I can login if I start the computer without the Dell D3100 plugged in. If it's plugged in, I get looped back to the login screen.

After I login, I can plug in the D3100 and connect to the provider and display on the internal display and one other monitor. I activated the second monitor and got kicked back to the login screen. Also, if I disconnect the D3100 cable without running the commands, it also kicks me back to a login screen.

Please let me know if logs are necessary - I can provide if instructed to. Thank you.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic

ungentilgarcon, 2018-11-30:
Well, I also have a boot loop with debian stretch with kernel 4.18 and nvidia video card 
Can't seem to make it work :/

ungentilgarcon, 2018-11-30:
though I tried this branch

AdnanHodzic, 2018-12-01:
@ungentilgarcon you resolved this part as part of: https://github.com/AdnanHodzic/displaylink-debian/issues/176 right?

If rest of you (with nvidia cards) try:

adding following to the `/etc/X11/xorg.conf.d/20-displaylink.conf` file:
```
Section "Device"
  Identifier "DisplayLink"
EndSection
```

and remove: `/usr/share/X11/xorg.conf.d/evdidevice.conf`. And please re-logging out into Xorg. If this ends up working for all you, I'll make new changes where when displaylink-debian detects you have Nvidia card that's settings it's going to setup automatically for you.

AdnanHodzic, 2018-12-04:
@ungentilgarcon In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

ttepatti, 2019-09-20:
Hello, I'm having this issue again on Debian Stretch with an Nvidia graphics card.

I had to uninstall the displaylink-debian package to even log back into my system so unfortunately I can't provide the output of 20-displaylink.conf, but I can provide the other three requested commands:

 lspci | grep -i vga
`00:02.0 VGA compatible controller: Intel  Corporation Device 5917 (rev 07)`

 lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
`i915`

 lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
`(this command provided no output)`

The output of displaylink-debian.sh --debug:

```--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 5917 (rev 07)
VGA (3D): NVIDIA Corporation Device 1d33 (rev a1)
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/70-synaptics.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
```