AdnanHodzic/displaylink-debian #121: DisplayLink fails to build in PureOS

miltonhowe, 2018-03-19:
I am running PureOS.

When I try to run the script I get the following message:
`ERROR (code 3): Failed to build evdi/1.0.335. Consult /var/lib/dkms/evdi/1.0.335/build/make.log for details..`

the make log contains several errors:

0.335/build/evdi_encoder.c
/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c:124:16: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .fb_create = evdi_fb_user_fb_create,
                ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c:124:16: note: (near initialization for ‘evdi_mode_funcs.fb_create’)
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:38:12: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .fault = evdi_gem_fault,
            ^~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:38:12: note: (near initialization for ‘evdi_gem_vm_ops.fault’)
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:60:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .unload = evdi_driver_unload,
             ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:60:13: note: (near initialization for ‘driver.unload’)
/var/lib/dkms/evdi/1.0.335/build/evdi_connector.c: In function ‘evdi_detect’:
/var/lib/dkms/evdi/1.0.335/build/evdi_connector.c:71:7: error: implicit declaration of function ‘drm_device_is_unplugged’; did you mean ‘drm_dev_is_unplugged’? [-Werror=implicit-function-decla
ration]
   if (drm_device_is_unplugged(connector->dev)) {
       ^~~~~~~~~~~~~~~~~~~~~~~
       drm_dev_is_unplugged
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c: In function ‘evdi_platform_probe’:
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:109:10: error: implicit declaration of function ‘drm_platform_init’; did you mean ‘early_platform_init’? [-Werror=implicit-function-declaration]
   return drm_platform_init(&driver, pdev);
          ^~~~~~~~~~~~~~~~~
          early_platform_init
/var/lib/dkms/evdi/1.0.335/build/evdi_connector.c: In function ‘evdi_connector_init’:
/var/lib/dkms/evdi/1.0.335/build/evdi_connector.c:154:47: error: ‘struct drm_mode_config’ has no member named ‘dirty_info_property’; did you mean ‘tile_property’?
                              dev->mode_config.dirty_info_property,
                                               ^~~~~~~~~~~~~~~~~~~
                                               tile_property
/var/lib/dkms/evdi/1.0.335/build/evdi_encoder.c: In function ‘evdi_encoder_init’:
/var/lib/dkms/evdi/1.0.335/build/evdi_encoder.c:72:12: error: too few arguments to function ‘drm_encoder_init’
   status = drm_encoder_init(dev, encoder, &evdi_enc_funcs, DRM_MODE_ENCODER_TMDS);
            ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-4.14.0-3-common/include/drm/drm_modeset_helper_vtables.h:33:0,
                 from /usr/src/linux-headers-4.14.0-3-common/include/drm/drm_crtc_helper.h:43,
                 from /var/lib/dkms/evdi/1.0.335/build/evdi_encoder.c:15:
/usr/src/linux-headers-4.14.0-3-common/include/drm/drm_encoder.h:178:5: note: declared here
 int drm_encoder_init(struct drm_device *dev,
     ^~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c: In function ‘evdi_modeset_init’:
/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c:145:3: error: implicit declaration of function ‘drm_mode_create_dirty_info_property’; did you mean ‘drm_mode_create_aspect_ratio_property’? [-We
rror=implicit-function-declaration]
   drm_mode_create_dirty_info_property(dev);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   drm_mode_create_aspect_ratio_property
/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c:148:3: error: too many arguments to function ‘drm_dev_set_unique’
   drm_dev_set_unique(dev, "%s", dev_name(dev->dev));
   ^~~~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-4.14.0-3-common/include/drm/drmP.h:76:0,
                 from /var/lib/dkms/evdi/1.0.335/build/evdi_modeset.c:14:
/usr/src/linux-headers-4.14.0-3-common/include/drm/drm_drv.h:636:5: note: declared here
 int drm_dev_set_unique(struct drm_device *dev, const char *name);
     ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c: In function ‘evdi_platform_remove’:
/var/lib/dkms/evdi/1.0.335/build/evdi_drv.c:117:3: error: implicit declaration of function ‘drm_unplug_dev’; did you mean ‘drm_put_dev’? [-Werror=implicit-function-declaration]
   drm_unplug_dev(drm_dev);
   ^~~~~~~~~~~~~~
   drm_put_dev
/usr/src/linux-headers-4.14.0-3-common/scripts/Makefile.build:319: recipe for target '/var/lib/dkms/evdi/1.0.335/build/evdi_connector.o' failed
make[4]: *** [/var/lib/dkms/evdi/1.0.335/build/evdi_connector.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-4.14.0-3-common/scripts/Makefile.build:319: recipe for target '/var/lib/dkms/evdi/1.0.335/build/evdi_encoder.o' failed
make[4]: *** [/var/lib/dkms/evdi/1.0.335/build/evdi_encoder.o] Error 1
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.14.0-3-common/scripts/Makefile.build:319: recipe for target '/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.o' failed
make[4]: *** [/var/lib/dkms/evdi/1.0.335/build/evdi_modeset.o] Error 1
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.14.0-3-common/scripts/Makefile.build:319: recipe for target '/var/lib/dkms/evdi/1.0.335/build/evdi_drv.o' failed
make[4]: *** [/var/lib/dkms/evdi/1.0.335/build/evdi_drv.o] Error 1
/usr/src/linux-headers-4.14.0-3-common/Makefile:1525: recipe for target '_module_/var/lib/dkms/evdi/1.0.335/build' failed
make[3]: *** [_module_/var/lib/dkms/evdi/1.0.335/build] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.0-3-amd64'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2

I have search for help but so far found nothing that explained my issue.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic

miltonhowe, 2018-03-26:
Hi there,
Thank you for your help.
I have pulled the recent file and can confirm that I no longer get those evdi build errors.
It does not however work, but I will open a new issue once I identify where the problem is.

AdnanHodzic, 2018-03-26:
Ok, sounds good and if you haven't please take a look at [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

miltonhowe, 2018-03-26:
I got it working and now have three lovely screens of linux on my desk.

FWIW, I had to do some intervention in order to get it to work. I had to add an `elif` for my distro (`PureOS green`). Then the displaylink script was crashing with the message:
`Unsatisfied dependencies. Missing component: Linux headers for running kernel`.
It is looking for a file named `/lib/modules/$KVER/build/Kconfig`, which apparently does not exist on my distro.

I modified your script to remove the cleanup so I could go into the displaylink install folder and disable that dependency check following the instructions [here](https://adangel.org/2016/09/26/display-link-dell-intel-debian/). And then ran `sudo ./displaylink-debian.sh`, rebooted and there was the service running.

I would do a PR to add my distro to your script but without understanding exactly why that other thing failed I don't think it would be appropriate to add support for the distro without resolving that problem. 

I have to write some little scripts like you suggested in the postinstall to simplify the screen configuration on reboots and unplugs.

AdnanHodzic, 2018-03-27:
If you could create a PR and I would review it and if necessary help you tailor it.

Regarding headers problem, I think it would be fixed by adding `PureOS` to line 273, i.e:

` if [ "$lsb" == "Debian" ] || [ "$lsb" == "Kali" ] || [ "$lsb" == "Deepin" ];`

Whole logic is in block under `# minor hack to alter the evdi kernel check on the fly, untill patch is merged into displaylink mainstream.` comment.

Otherwise create a "feature request" to add support for PureOS and I'll do it when I have time.