AdnanHodzic/displaylink-debian #122: Doesn't work properly on Ubuntu 18.04

RichardSouta, 2018-03-20:
Hello, 
I used your script on Ubuntu 18.04, but when I restart my PC, I cannot even log in with USB connected to dock with second monitor stucked in.
Before rebooting, which causes issue described above, I see grey box under cursor and when I highlight text with cursor, I don't see all the words highlighted.

Please contact me for more details if needed.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which should resolve your problem.

Closing the issue, if problem persists after running re-install with latest changes please free to re-open this issue.

closed 2018-03-26 by AdnanHodzic

milsanore, 2018-09-04:
I have the same issue with ubuntu 18.04.1, but only with proprietary nvidia drivers, so its ok.