AdnanHodzic/displaylink-debian #124: Works great on Debian Stretch! Just got a little lost for a second in the readme...

finetype, 2018-04-14:
Just thought it might be worth updating your post installation guide for clarity. I am just super grateful it exists, as does this whole project. On my last machine this didn't quite work for me, but you made an update shortly afterwards that supposedly would have solved my issue. I'm on a new machine and running Debian 9 now, gave it another shot, and it works perfectly. Thanks!

That said, not seeing anything after reboot was a little disconcerting.

Luckily, punching in some commands, presumably the xrandr lines, did the trick. Still, it silently connected it in the 'off' state, and it wasn't until I went to the displays setting manager again that I now saw it connected, just turned off (whereas every monitor I've ever connected has always just automatically been 'on').

Is there an easy way to update this software to automatically connect it on boot up, like any other monitor? I'm mostly a web developer, but I've written some bash scripts before. If something like that would be welcome and you know the best route to go down that line, I'd be glad to take a stab at it, though I can't make promises about the quality of my bash code.

Anyways, thanks again. :)

kopax, 2018-09-13:
Hi ! You don't have a frozen screen with part of the display?
Does it display directly full screen? What screen do you have.

I have a Lenovo X1 Carbon gen1 with a fresh debian stretch, I have issues https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-421037991

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic

kylebakerio, 2019-01-22:
@kopax, no, it worked perfectly for me. Thinkpad T460.