AdnanHodzic/displaylink-debian #125: Install old driver version

pa102, 2018-04-16:
I ran the script this morning in order to update the DisplayLink driver to the latest version (4.1.9 -> 4.2).

Following the update I have had numerous issues so rolled back to the old driver (4.1.9).

To do this I first ran the script and used it to uninstall. Once that was done (due to my extremely limited bash skills) I butchered the commands on lines 14 (version=) and 16 (dlurl=) to ensure it didn't return the latest version.

Would it be possible to request a feature that allows the user to select which driver version gets installed?

Thanks for your help.

AdnanHodzic, 2018-09-12:
@willselby why did you close this issue? I think it's a good idea, to have ability to install different driver versions. I just haven't had time to look into it. 

I also added `[feature request` prefix to it, so I have better overview.

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

While working on adding support for 5.1, I've realized that this is not a good idea to give users use older/different version of the driver. As DisplayLink's case "newer is always better" statement is definitely true.

closed 2019-02-20 by AdnanHodzic