AdnanHodzic/displaylink-debian #129: Error on boot, system freezes

sakrow, 2018-05-28:
I have Dell XPS 15 9560, everything worked fine until I run `displaylink-debian` and reboot.

Now the system freezes in this screen and not boot.

[image]

To run Ubuntu 18.04 was necessary to modify kernel run options adding `nomodeset`

After, I install nvidia-384 drivers. Then I run `displaylink-debian` and reboot. Now i have this screen.

Something to do? Any Workaround to work?

AdnanHodzic, 2018-05-28:
Reason for this is because `displaylink-debian` by default [will enable "modesetting"](https://github.com/AdnanHodzic/displaylink-debian/commit/83a6587d4fa85086c0c2f54d76e3f051831ee8ed). 

Changes need to be made so `nomodeset` are set used with >= Ubuntu 18.04. I'll do this as soon as I have free time.

sakrow, 2018-05-28:
Ok, thanks!

I renamed `/usr/share/X11/xorg.conf.d/20-displaylink.conf` to `/usr/share/X11/xorg.conf.d/20-displaylink.conf.old`

And now the system boots normally.

Is necessary to do something more?

Thanks for all.

sakrow, 2018-06-05:
I tried with more file options, and nothing works.

Besides, the pointer blinks in all monitors and in sometimes displays black area around her or not updates the background.

Anyone know any solution?

AdnanHodzic, 2018-11-24:
I think issue is related to: https://github.com/AdnanHodzic/displaylink-debian/issues/149

Could you please tell me if you use any custom X configs, and can you you give me back output of these two commands:

`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`

Also I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic

hagen-thorn, 2020-03-26:
I have the same issue: system does not boot, got stuck after the DisplayLink service has started.
I was able to boot in rescue mode, text only.
I was looking at my xorg configuration, but I do not have 20-displaylink.conf at all. Moreover, I suspect that the script has deleted some of my display configuration, because in the Xorg logs I can see it says that a display is detected but has no corresponding config record.

Is there a way to restore my previous configuration? 
Or maybe I have to delete the DisplayLink service?
I need to clean up this mess, to be able to boot as before, even without USB displays. Basically, I need to revert everything.

I tried to run the displaylink-debian.sh script to uninstall, but the script does not run in text only mode, saying "Can't open display".

Please, help!