AdnanHodzic/displaylink-debian #130: Debian SID : Fail to connect screens

odivel, 2018-05-28:
Hi,

This morning I had an issue when trying to connect my displaylink device.
When I activate my screens it brings back to the login screen.

I thought it was kernel related (this weekend I updated to 4.16) so I reverted back to 4.15.0.3 and reinstalled the displaylink driver but I have the same issue.

Can you help me with this ?

Regards,
ODI

AdnanHodzic, 2018-05-28:
Are you logging into X11 session? WIth Sid/Booster Wayland has been set to be default, and Displaylink doesn't work on Wayland ATM.

odivel, 2018-05-28:
Hi,

I'm aware of that and I'm logging into X11 session (choose Gnome on Xorg when logging)
It was working fine until today.

The only thing I've done is an aptitude upgrade, kernel was updated to 4.16 but I have the same issue with 4.15.

odivel, 2018-05-28:
to add more informations 

dlm.service is started 

xrandr --listproviders
Providers: number : 3
Provider 0: id: 0xb6 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x81 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x41 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

And you will find in attachment a dmesg

[dmesg.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2045247/dmesg.txt)

AdnanHodzic, 2018-05-28:
What happens if you temporarily remove this file: `/usr/share/X11/xorg.conf.d/20-displaylink.conf` and reboot?

If doesn't do anything, could you please add `nomodset` parameter to your kernel?

odivel, 2018-05-28:
Ok so I tried to remove /usr/share/X11/xorg.conf.d/20-displaylink.conf but it does nothing

And when adding nomodset nothing too (tried with nomodeset and it's not booting)

Here is what I added : 
GRUB_CMDLINE_LINUX_DEFAULT="elevator=deadline radeon.modeset=0 modprobe.blacklist=radeon nomodeset quiet"

Maybe I've done something wrong ?

I tried with Kernel 4.16, should I retry with 4.15 ?

AdnanHodzic, 2018-05-29:
So after adding `nomodeset` it's still not working? Also why did you add rest of the parameters, i.e:

`elevator=deadline radeon.modeset=0 modprobe.blacklist=radeon`?

odivel, 2018-05-29:
After adding nomodeset it's not booting at all, I have a black screen and nothing (have to boot in recovery mode to remove nomodeset)

I added elevator=deadline to change the I/O scheduler, and radeon.modeset=0 modprobe.blacklist=radeon is a workaround to solve a vce init error radeon 0000:08:00.0: failed VCE resume (-110)
Just tried without these options and with Kernel 4.15 in case of but same issue.

odivel, 2018-05-29:
If I cannot solve this issue I think I will just migrate to Ubuntu for my professional Laptop

Too much issues like that with Debian SID, maybe I've underestimated the unstability of Debian Unstable (Yeah I know many people would say that they never add problems for years)

AdnanHodzic, 2018-05-29:
I'm really not sure what the problem is here.

It definitely takes a lot of dedication to use Debian Sid as your work laptop. I would recommend using Testing/Buster, or even if Stable/Stretch with APT pinning. 

I wrote about this topic few years ago in case you decide to go down this path: [Taking control over Debian and its package repositories](https://foolcontrol.org/?p=1708)

Either way, let me know if you end up having same issue with whatever out of these you decide to go with. Similiar problem was reported on [Ubuntu 18.04 yesterday](https://github.com/AdnanHodzic/displaylink-debian/issues/129) but was resolved with `nomodset`

odivel, 2018-05-29:
Ok I think I will make a try with Ubuntu 18.04 and report if it work well

Thanks for your time

odivel, 2018-05-30:
Ok so I moved to Ubuntu 18.04 and it's working fine

For now I will keep Ubuntu and switch to Debian later

AdnanHodzic, 2018-05-30:
Great, also out of curiosity did you have to make any customizations. Or `displaylink-debian` install got everything working out of box?

odivel, 2018-05-30:
It worked out of the box,

But I think I will go back to Debian, I will try testing with apt pining

cilynx, 2018-06-01:
I can second the issue, but I haven't figured out the cause yet.  I've been successfully running a pair of displaylink adapters under Sid with 4.16 for a while and all of a sudden today while doing nothing out of the ordinary my session crashed and it kicked me out to the login screen.  I haven't run an `aptitude update` in several days.  When I logged back in and went through the normal dance, I can set things up through `xrandr` fine, but once I enable either display in `gnome-control-center`, it kicks me back out to the login screen.  I'll post back if I can figure out what's going on.

redjamesg, 2018-06-01:
I too noticed this, using the latest install script, the only temporary solution I have found is to remove the /use/share/X11/xorg.config.d/20-xorg.displaylink.conf file and keep my old one that I made alomg with the very first install of this driver.

cilynx, 2018-06-01:
I managed to get back up and running by `sudo apt purge gnome* xorg*`, reconfiguring networking from the cli, swapping `unstable` to `testing` in my `sources.list` then `sudo apt install gnome xorg`.  I'm still running the newest 4.16, so the breaking change must have come from an update in `xorg`.  I'm guessing my "random" crash came from whatever part of xorg got updated with the breaking change finally being reloaded from disk on my machine.

@redjamesg -- can you share your `20-xorg.displaylink.conf` that is working for you in sid?  If I can spare another couple hours to trash my machine again tomorrow, I'd like to try it out.

odivel, 2018-06-01:
Hi, 

Ok so I decided to reinstall Debian testing and it's working fine.

I'm using apt pining for some softwares but displaylink is working with the last script

redjamesg, 2018-06-01:
This is my current xorg.conf 
`
Section "Device"
        Identifier "Intel"
        Driver "intel"
        Option "VSync" "false"
EndSection
`

AdnanHodzic, 2018-06-01:
@redjamesg what happens for you if you add `Option "VSync" "false"` to `/usr/share/X11/xorg.conf.d/20-displaylink.conf` file?

Would love to extend it so we could potentially have an universal file which we could use everywhere.

redjamesg, 2018-06-01:
What I was experiencing, was after installing with teh latest script my three monitors were working as expected without any issues, but two days later they startet acting up.
Occationaly they would flicker and drop me to the login prompt of my GUI (xfce) once again but after a few activations they would continue to work.
This tendency would increase in frequence and at one point I was unable to activate the monitors, they would get stuck in the state described in #68 and since I remembered my fix for that I tried to change the 20-displaylink.conf file to just being what I used then, and now everything works.

I will on a later instance try the "VSync" "False" setting in the orriginal 20-displayling.conf to verify if that has an impact on my issues.

rosetree, 2018-07-28:
Had a similar problem when using this on Debian Testing few months ago. I followed the advices in #68. My `xrandr` didn’t show any providers (as [described in the Arch Forum](https://bbs.archlinux.org/viewtopic.php?id=214939)). The monitor worked, after I disabled the Wayland backend in GDM (see [Arch Wiki, GDM “Use Xorg backend”](https://wiki.archlinux.org/index.php/GDM#Use_Xorg_backend)).

AdnanHodzic, 2018-11-24:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic