AdnanHodzic/displaylink-debian #131: Conflicts with Logitech Unity USB

elimtaft, 2018-05-29:
Thanks for this great script.  I've been wrestling for two days to figure out why it was working for a second, and then stopped working.  I was reading through all the trouble-shooting issues and related issues and came across someone who mentioned a conflict with other USB devices.  I thought, surely this can't be my issue, but why not try it?  I literally unplugged my Logitech Unity USB device, then typed "systemctl restart dlm.service", and the monitor connected to the DisplayLink adapter turned on.

Here's my setup is:

Lenovo X1 Carbon 5th Generation
Debian Stretch fresh install
1 Acer S271HL DBID 27-Inch Screen LCD Monitor plugged in via HDMI directly
1 Acer S271HL DBID 27-Inch Screen LCD Monitor plugged in via DisplayLink USB to HDMI adapter

It turns out I didn't have to do any of the additional trouble-shooting stuff.  This is literally all I had to do:

- Plug in one monitor to the HDMI port directly.
- Disable laptop's LCD screen (might not have been necessary)
- Install using the displaylink-debian.sh script
- Pay attention to the messages from the script, which first told me to reboot because evdi was already in use, then tried again, where it told me it was successful and I needed to reboot again.
- Plug in the DisplayLink USB to HDMI adapter with the monitor connected to it.
- Unplug my Logitech Unity USB device
- Type "systemctl restart dlm.service"
- Monitor found.  After setting up the positions to my liking with 'gnome-control-center display', I plugged my Logitech Unity USB device back in, and everything seems to work.

So I'm guessing, when I go to work today and come back home, and power things up again, if I have the same problem with the second monitor attached to the adapter not turning on, I'll just have to do the last three steps above.

In case it's helpful, upon request, I can post some info from /var/log/messages.  Also, I have two of these DisplayLink adapters, and my preference would be to have both monitors connected to them.  That way, I can try to use this Pluggable 7-port USB hub.  I'm just picky that way... I'd rather plug in one usb cable and have that connect everything else, rather than have to plug in one HDMI, one USB for the second monitor, and then another USB for my wireless keyboard and mouse.  But I might be asking too much :)

closed 2018-08-22 by elimtaft

AdnanHodzic, 2018-08-22:
@elimtaft I see you closed the issue, do you mind telling me what happened in the end?

elimtaft, 2018-08-22:
I wasn't able to resolve the issue on my own, and eventually I wasn't able to get both monitors to display even when I wasn't using the Unity USB stuff.  So I caved and bought the Lenovo Thunderbolt docking station, and that worked without having to install any additional drivers.