AdnanHodzic/displaylink-debian #132: Hint for 640x480 issue on Ubuntu 18.04?

MichaelJCole, 2018-05-29:
Hi, I'm having trouble with the DisplayLink driver and they haven't released an 18.04 version yet.

My displays are stuck at 640x480.  I documented the issue here:

https://displaylink.org/forum/showthread.php?t=66106
https://github.com/DisplayLink/evdi/issues/129

I'm experiencing this with the regular DisplayLink installer, and this one.  

Have you run into an issue like this?  Did you find a fix?

Thanks!

MichaelJCole, 2018-05-29:
Hi Adnan, I tried this a couple different ways and my laptop wouldn't boot with nomodeset.   On boot, it asks for the password to decrypt the drive, but froze before getting to the login manager screen.  

Is it working for other people?  Any other suggestions?

I'm thinking about giving up on ubuntu 18.04 and installing a new distro.  I have a 2nd drive encrypted with /etc/crypttab  Any suggestions on what's a working config?  

- Debian 9 Stable? Testing?
- Linux Mint? 
- Ubuntu Gnome 16.04 (last known working config)

Thanks!

AdnanHodzic, 2018-05-30:
Honestly, Ubuntu 18.04 should work fine. Could just please try `mv /usr/share/X11/xorg.conf.d/20-displaylink.conf /usr/share/X11/xorg.conf.d/20-displaylink.conf.old`, reboot and see if that does anything?

There was a same report like this couple of days ago and person got it working with what I mentioned here: https://github.com/AdnanHodzic/displaylink-debian/issues/129

MichaelJCole, 2018-05-30:
Hi Adnan, I tried that but still the same result - the gdm login manager never shows.

[image]

MichaelJCole, 2018-05-31:
Hi Adnan, the nomodeset was a boondoggle.

The hardware has DisplayPort and HDMI ports for each of the monitors.  Replacing the DisplayPort cables with HDMI cables fixed the problem.  It's astronomical to have two cables fail at the same time, so It seems like a driver or firmware bug.  

After reinstalling different versions of linux about 6 times, I'm back to Ubuntu 18.04, using this repo's script to install it.

AdnanHodzic, 2018-05-31:
I agree that's too much of a coincidence. 

Either way, thanks for reporting back, I'm glad you manage to get it sorted out. I will close this issue, please feel to re-open if anything similar pops up again.

closed 2018-05-31 by AdnanHodzic