AdnanHodzic/displaylink-debian #134: no lightdm on xubuntu 18.04 LTS

imthenachoman, 2018-06-23:
I was able to install the DisplayLink drivers using your script. However, when I reboot, lightdm does not start. I cannot figure out why.

`systemctl status dlm.service` shows as running. 

X does seem to work because I get the XFCE loading screen on boot but then it never enters lightdm. It just sits at a black screen with a blinking `_`.

Not sure where to look to troubleshoot. Any assistance is greatly appreciated. 

Thank you!

asmith26, 2018-07-19:
Hi @imthenachoman 

Was lightdm starting prior to installing this? I was finding my GUI/desktop was slow to start (sometimes taking 5+ mins to start). Following [this](https://unix.stackexchange.com/questions/455104/lightdm-slow-to-start-linux-mint-19-running-ubuntu-18-04), downgrading my kernel version to **4.15.0-23** (from -24) worked.

Hope this helps!

imthenachoman, 2018-07-20:
Yes lightdm started fine before it. In fact, if I uninstall the DisplayLink drivers it works again. 

With that said, I bit the bullet and got a new video card with multiple outs. :)

AdnanHodzic, 2018-07-20:
Glad you managed to solve your problem, un/install can solve a lot of problems, and should probably be added to [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md).

closed 2018-07-20 by AdnanHodzic