AdnanHodzic/displaylink-debian #135: Can't set provider output source

Fadarrizz, 2018-06-30:
Hello!

I had displaylink working for quite some time. But recently it just stopped working. I tried to install displaylink again via the steps on their site, but the installation stops after:
>adding udev for DisplayLink DL-3xxx/5xxx devices

So, I am now trying to install displaylink-debian.sh

After installation, I am trying to set provider output source. After listing the providers, I got:

> Providers: number : 2
Provider 0: id: 0x64 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
Provider 1: id: 0x3f cap: 0x4, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:modesetting

Then when I execute `xrandr -- setprovideroutputsource 1 0`, I get:
>X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x3f
  Serial number of failed request:  16
  Current serial number in output stream:  17

I guess it has something to do with the 0 outputs and 0 associated providers.
Does anyone know how I can fix this?

Many thanks!

redjamesg, 2018-06-30:
It seems like there's a space between -- and setprovideroutputsource, they should be combined.

Fadarrizz, 2018-06-30:
Thanks for your reply.
Yes, that is true. I just typed it myself for this post, not copied it.

But I've crashed my system, so I had to do a fresh install of ubuntu.
After the installation, I tried the steps from Displaylink themselves again, and it worked!

closed 2018-06-30 by Fadarrizz