AdnanHodzic/displaylink-debian #136: Stuck on Building EVDI

codercodingthecode, 2018-07-02:
It just hangs in there for as long as you can let it be.
Ubuntu Budgie 18.04 Kernel 4.15.0.24.
I tried  the master and  modesetting with no avail.

[image]

codercodingthecode, 2018-07-02:
Alright, so I solved it.  
I did take a few approaches, so kinda hard to pinpoint which one.

I had secured boot turned on. That was the only way I could get the boot locked into the uefi/biios

I turned off secure boot and ran the original installer. It worked straight out after that.

I also did this, so, after rebooting Im not sure if it is needed, but I will leave here if someone still get stuck. The selected answer might've helped.

https://askubuntu.com/questions/762254/why-do-i-get-required-key-not-available-when-install-3rd-party-kernel-modules

AdnanHodzic, 2018-07-02:
Great, thanks for the follow up.

closed 2018-07-02 by AdnanHodzic