AdnanHodzic/displaylink-debian #137: no monitors detected on Ubuntu 18.04

nazgee, 2018-07-06:
Installed fine and runs ok:
```bash
stawjmic@ul001743:~/workspace/displaylink-debian$ systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-07-06 13:24:25 CEST; 5min ago
  Process: 1643 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.2.29 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1646 (DisplayLinkMana)
    Tasks: 69 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─1646 /opt/displaylink/DisplayLinkManager
lip 06 13:24:25 ul001743 systemd[1]: Starting DisplayLink Manager Service...
lip 06 13:24:25 ul001743 systemd[1]: Started DisplayLink Manager Service.
```
But the list of providers is 0 (2 monitors connected)
```bash
stawjmic@ul001743:~/workspace/displaylink-debian$ xrandr --listproviders
Providers: number : 0
```

marcelotrevisani, 2018-07-21:
The same with me
Did you find a solution for that?

Azza007, 2018-08-16:
This is the same issue I’m having.

azalucha, 2018-08-28:
Similar 2 monitors connected, service is running, doesn't detect monitors:

>> systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-08-28 10:07:01 MDT; 1min 26s ago
  Process: 1010 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.2.29 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1014 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─1014 /opt/displaylink/DisplayLinkManager

Aug 28 10:07:01 calculon systemd[1]: Starting DisplayLink Manager Service...
Aug 28 10:07:01 calculon systemd[1]: Started DisplayLink Manager Service.

>> xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting

ak9999, 2018-09-28:
I'm having the same issue, but I checked the log in

> /var/log/displaylink/DisplayLinkManager.log

and got the following output:

> UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADc3SFuE3wL+QT3GNZJWHAGbuEdinxloggMORGnbuQ6SZSa31gNy0HO2ze7bd0TLJa3BNtmzN/u40gehGsXFn3/Ve
ZjkII9ANA26kQnxsGo1dnnUUMkMT9XrDIfj9Xb+6TZNiJYl7+qD6xqQRdac61izYNQcOxIa7xwxdXqjWnEMUmBDBE7R1wxNRTSLvMAo6itab8yZnjF1OAzsesXfeWIFgz/yrHNsgeQ5JhFuitUsa9n
GtBLVrYpJaqnEQQtHdg2Uez09d89YDa3tq+afeVxC9Bgul6ga1W5FrwnZLgNmeaj+bOnLU3y/za1L6dwuUh6IqakNyx3vTrb0DDnb2fw7w==
F6ny/x7M/S/F1fu0oS2K41aR1n8URVQQ+ipS4NMLIkQoWhR5YEQMq2Tq5SP+RdIHK9blfFEBFZNJ9hcJkBt+9QJprvq+g0Gm3owr8jGfZ85yh8s+ChwENKeJRiQgs/Wr5n6JOH9b3EEAIUxPzTQ2/W
zFvpDR28TN9BqCUn0QUts1lnKX3+FfOC4J4lZhnsLfTKAkGxeNSaZ/2jtwMeX1VRVY8dyu3l8hVfogXC3sVxD5qwjaOd0i3moJWwbyDWQspunIY2GAONcgkH+6u87UPjHMuZcfCGPyu/YKpcEZg8FP
DLw58uU0o4Upbtq3aGaaOOV475tZ7Gi3zXJRBnfi+A==
1G2JyAkrPgv/ueMVOOIxM04Oqx1yXoXoCv99KNN0AqhDacpn6iaXn6wmdfDooqxV7oThGMzwPuFrtYSzyfq6xE6WsLuMnBwxjlKvfTr7/JQus3ALYXD05D+XjRIP2EHKo19rLUmtkk+hfGeuh/OSEc
V1+pZ9fSpkkS7+KvVF4Q2HkJRjLnWxqoRapd++vGXFaIOGKlF61VjIUNWo0RWRDINJExW3ONMVXatrNwIvW20zgB05STnpz9fMYGq4oCKUHU0AhXZ8sd4RVw03fYnTjC5dtOCZ3WuHmEebgLfvVO6R
1ZEvzsac1kFKCD9xF5EYqd1WTqIHrWt09zv2OJAHmQ==
UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADcQJ5F58oSwdkwwMvFAACG0oiOQbhhzIzsewqLk2ZGrYyQa6eoGs+fJSwqa6cjZoI7rppElkRpR0zbKaSg9bTHrn
cHCXq2xqz3JjH9TIGRm52vxIIvkCs1JrrZGX0XatGfDDKw+5ehQK31GePWeZVpu08U+Ki/OdZGCXkfcXJ5XivzOHMv/h+xQh5inFy9gY8wNcvzt1hSvoEtybRyvOYF+hngiY0Vo/EPisxYlw9J1+xG
dF5qzQYj9Nj67ieJv9wD/fumBD65RROU9va6wPHEL+uVM5Hv6niCut8/d/msMF3R79qCxh/5f+mZbAtV4Hkm7nfaBVWIjlcKe7kjFgCzAQ==
F6ny/x7M/S/F1fu0oS2K48YS/f/bZvkfgmUgnVDdTNYtYE2f39eZZDQcJwGd8PVFEUWqMo6dCPfGqw/yaDZjjesytU8vpcMGP9sjx8JdSA63zye1ONBqKJhpYgF9uC5rxv3DD1iVYXP0Ugjk4Q5N+O
wXVia1x5f96dncWliXYJ30y6zQbDBLnTJSeZ/koxWr72jjYrNqZTVcfxTsbtvVa2a5YcA01cna9Dk2331X9EYgfDsHnecb/vvhFUEMDKA/YDopLWRCmUC4/FiZ0fN7AHYZcQwGCTEdjrB+WQoheNmy
bHejpwPj7QQirA+PEHEVIW6TMweDt0bUUq3lnX+sxg==
1G2JyAkrPgv/ueMVOOIxM/6hX8Brqp3ECNCyeXTPqhDXqDisLHWgVatWSYt6p3w6uXur0eh2OKr2OCSk9wIbw1H8auT66KEU/tJjteSOzlMY1LTC5JWhwtDTMcukRjMITQezogkKwW49/BDosuIdqe
aAHQfZz9yPNDUh1bULif+vgtw6kzTDGl1lI56gdeLB7FK34YNlQx0EwHZIyweaqrdlMswiIRJKx6harTHxMRm3l7a2J1ggH+Hz8Pt71uAkKt0xL594W4yAwKGX0yXKgbaiyEuND2Fx3Zth/XJawh27
wlDM1PJd5B7dK6b9yhtZmNoGZKvdOB2wrRMoLqOBzg==
UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADcxW/0yE9suS5f5ATuNWiD7YwYsbeWQ7hjxv4hSl9ZLUBfXKQcrqSGl72q7bZC4Z1kNxi45xndZG8nuNc2mLvvB/
PeQF8sZ+qld1A5rtuqgZQN04Szxr7DQHvOHTFlAxa1MlS30ktHGtjAkP0z3lqRcA66KKbLPg5FBe6R+pGIJDrXbi7JWo6fh6L15nt++sTUbWq4/RpmOLnfviL5fVYZUdbVyWNQm+b08opBfpVctHhV
l8wubmhEG1jpXhvLuttdX0gicBad937oGiD6lxCjlmWq44LQnOqk8IPdO3NWzFLq+6lGdIVJ/yadnlXVYMCSI3vlRsps6TEvT6nAtyAnzQ==
F6ny/x7M/S/F1fu0oS2K4x4ImfNHPWiWcLlu4cJZn+85rxueUcxUBZWOQIK9qYpKz3mN2hz6mvRRMLhNuKcvgKggILceoD0n+qzFLYR6wLsbAgmfgWuQ70sPhEouRBoqQggxmfDYL3PmCIxDvKR3J0
cfiH/UvxVSbnxe1s233gEqU/sRXRzWzboe0E5QcXwI/CAG7QCeOdw6fCuQ4832JYw7krG0tD249P3XZSSjg2j4DSckPyiTYrICrzanjFX2zhf54CEgECqRH7aC28xETJczR0HS/FLIihGhlqL1/40B
vvTbLQueiTtnJCIJdiqLiTts+1wAfYPUKDHScktcAA==
UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADc2tiNInzxwcpn3poxXPE1QRGTOqZj0VO7ifsztkEuQ81BdxG2m7avA2VhOKrqjbLJUoxkF+yjSQ0ZC6Ff2uViaP
7Fcb3xhCT1EeJN2PnV2Ru/RIjLtBtv54DTYLJv1j92fr8zZDDSbcWKu5TXcr57E7ZmVTMeR719mjfnKRz/3W4pz8+gWM/ytaHjnrvChRtPgu8/8PswOd8dxvarcaD1Q728AvcOV5N53pJCe6QXSL0Z
CC5NLOEg+YVkDdKKWJhjSo8xiA/e1RF826pWCd0j+Dr98TnQLPOgv5Rt9CEPVFCFHHoo0avHuSLRrKvcNF79mXj/KKABJxoCxCjljVj84Q==
F6ny/x7M/S/F1fu0oS2K48YS/f/bZvkfgmUgnVDdTNYtYE2f39eZZDQcJwGd8PVFresHwbPJXICsN+vmQEs5TQoRdFjddC9fsnGu8bX7IpeefH3Ak2ZpP/a0QQoQPwOWl7BLqszMJyVdwlYJBCrybx
dZam2XmUevT3qmmQITMKTTipi9J0nTa0jjXmZZoOu8Ov802pGwXNB7sDL0o6miwVgBNnLjwBJ5u6TBAsP6VhdiiNra4ni2ZeAaAuEilq/vXRQchwb8fIZiUI14Nr9hMlmh0bl0yW5gQTB4APo691W6
tYcafeSvH0C3i3ZwgNOijnPE9Fru3vzyv6CNiLJMKg==

I have no idea what this means. Can someone else post their log?

dhong44, 2018-10-07:
I had this error recently and realised the problem was I was running 18.04 with wayland. Just had to switch to X on the login screen and everything works great now.

JoranDox, 2018-10-10:
@dhong44 which X? I have:

- "ubuntu" (assuming this is on X?)
- "ubuntu on wayland (this is wayland != X)
- "Gnome" (assuming this is wayland?)
- "Gnome on Xorg" (Xorg == X ?)
- unity (?)
- weston (this is the reference wayland compositor iirc? so wayland?)

every assumed X that I've tried gives me (with [these methods](https://unix.stackexchange.com/questions/202891/how-to-know-whether-wayland-or-x11-is-being-used))  the idea that I'm still on wayland, maybe I need to reboot in between?

azalucha, 2018-10-10:
@dhong44 Which X do you mean?

azalucha, 2018-10-10:
@ak9999 Similar nonsense:

UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADceTc5bThyKaiRwkwSSzXPjisqkF/j6oIHc6ZY8dKBvbruIM1pq2HR1NtnlCLeyfLULGWDhEWJVt1zlXcxXbXqwgGks2UDj2xCRGXg487JO81/5wFw5YA0EhrPQkv1rGbMPD+hQ/nE7i9NX49ShkOrsxOXr00dzJDAXlfWnYpPHejaTGgsH2mETwpK2VY8nQpxserBwM6v9OkTCpMK5zYKyFbfIvVLH7TABBRdYdiFa36j1uXrKh7keVRB0RGYK39Oxd2ICxwEG8DvwfjBEHSgQHhGIa3GkaE4n1cvxBVifzzrvByMH2jPii9FaBrEX3xx0aYBlIfctJIwNgQq98Rn9A==
F6ny/x7M/S/F1fu0oS2K41aR1n8URVQQ+ipS4NMLIkQoWhR5YEQMq2Tq5SP+RdIHlQTLs6WnGBR8YVIpMuXirDckyIRknGS/xIsvxCSNLkZG50+cjIuI9CuXo+Jh66gLecYAAkaFbSiy7vL4LVW0TlzxvazWuT3Q/RSA9Rk9peuwbBqk4GiS4zKtetfVDb9pGhth6l/6SOt7LtzZYGYh6wypqzATK7z0UgN7HBRzLUSAURxnnPc1qYh0JnMBua3fYx7PZwUBRx0DvX56R/yK8TQa9cCIinYOIxXNO5M8GYWrO38fFx6Lj0HTCg0cN6OiLCdBfVQLAkrppTpN/t+xCQ==
1G2JyAkrPgv/ueMVOOIxM5SCZGe4erbs/evVIzkH/gLRDj2rMhIPWHl+7q679eu+mnKGVjz0Rve0tD92QNtSnFXUkHs6iFKIFj4S2h85L9msj/9w8PJVQWEwk/swt3C5Xiqmc9lYUssSACT7Qy6Cyu/3vftJ1RRzJsgVFn18VPPey8puTjDTjdQJrObooCs5GdBWLK5W6z5324AkkNrDYfHddc8AsHWHKJk4rNv99+dSVjTQKmKbqlhl0DwrlfHp8zrP+cl3Kb/vob1dW7f6D/jcUPDUQ3H+tZ5evPAQa9KNS0czxhSnkiOf9Yc5hjLE2/Q0NlR183YdMInzgDJJpA==
UVLY+3xs9LIslFFJL9zz9ZX/E8tch0h1gXBEL4FgoejQ9yQ/0abH8BYtlJeU1ADcwOh6ZcsHYfpIN6aKONeTeiADz8mF2ZPFqih5ekFJpif9KAtB7e5N/48EdSR9Q5i8cqTEIi5aNnPs8jsUbG9AZFDbkI1aHl1PUD8tip0K/kgHLhkX1yllU4Vp9Tm3Dr4xKhawl/c3lOOiQvYhJaUUHIcn25Pxd8jq6UtWLQKUePqNWrAd+7giMPKgBbneyPYVA+8m/6ysdZFmJvK2HDvjnoXnM4tKmukXiTGzCyG8tL2QPB64M4t5FLeAes0/HaPPoKmqDQPSD9YphO8HiDo50oNATYpgj+BbhIxKmvq854LJ77rgqWF/+PViQ9Itt66XQ3U0hul1fMepGvyN6OuAfQ==
F6ny/x7M/S/F1fu0oS2K41aR1n8URVQQ+ipS4NMLIkQoWhR5YEQMq2Tq5SP+RdIH6rBwqI0Vwuy9StFKkLv2NLBifBsSXmm4y5Cs0kkK299bTXXmd8znpehod8Lq27fQBRghRBcI9aDGn8WbcTm56nMiHJjJhLlKezC1u/ZweKT63//xLdq7j8l/bNpn9SilHZSJ0Q+TVkS+pqgOGbnouLYYANHYJOovAF26uMx1j/vA4JhFPFjv4g0oC9aS5FqIxW/TQb0Rx7LVvYMu2ib76aZHVU2ZKqss/urnb7ZeBcd8U59KnXUITd6d09V6smc9t7gLDamawXMAvmNTfCkW+w==

dhong44, 2018-10-10:
My options when logging into Ubuntu are 

- Ubuntu
- Ubuntu on wayland

If I pick wayland, the monitors aren't found but if I use Ubuntu they are. 

My log, seems to have similar output to what you are seeing.
[log.txt](https://github.com/AdnanHodzic/displaylink-debian/files/2465529/log.txt)

JoranDox, 2018-10-11:
I got one screen working (but I've got two connected so this isn't the whole solution), my issue was that while it pretended to start X, it really didn't, and probably fell back on wayland regardless. Reinstalling X solved that part for me.

1. I reinstalled X following these instructions:
    https://ubuntuforums.org/showthread.php?t=1150319&page=2&p=9496946#post9496946
2. apparenly the `nvidia-config` part fucked me over because at that point one external screen worked but the resolution for both that screen and my laptop screen got stuck at < 1000 x <1000, deleting (or moving) /etc/X11/xorg.conf worked (you might need to do this even if you don't run the nvidia-config part, no idea)

So now I'm halfway there, but if something comes up that fixes the detection of my other screen, let me know.

BlueSoD, 2018-11-23:
The Solution from #172 worked for me, i am running Ubuntu 18.04 on a HP EliteBook 840 G5
had the exact same problem u guys have.
Linux Kernel:
4.15.0-39-generic

BlueSoD, 2018-11-23:
@AdnanHodzic here u go :D
username@computername:~$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
username@computername:~$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915

kopax, 2018-11-23:
Hi @AdnanHodzic, sorry to ask again but you have still no idea about my Asus screen? Should I film the error?

AdnanHodzic, 2018-11-24:
Also, could everyone in this issue please verify if they went through [post installation guide](https://github.com/AdnanHodzic/displaylink-debian)?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic

Zer07, 2019-02-04:
I am having the same issue. I started with the official displaylink driver, but that would stop working after rebooting. Then I found this driver and it doesn't see the monitors. I am connected through Plugable UD 3900 dock.  

How do I uninstall this thing and reinstall it? couldn't find the instructions on readme

Sorry, new to linux, here

Zer07, 2019-02-04:
I followed every step in the post installation guide. Everything is good till i get to this step: 

Setting provider sources

    Check providers, i.e:

    xrandr --listproviders

If you get a list of more then one provider, it means your displays were detected.

    Set provider sources, i.e:

    xrandr --setprovideroutputsource 1 0

    xrandr --setprovideroutputsource 2 0

AdnanHodzic, 2019-02-04:
@Zer07 contrary to what you state, it's very clear [how to run `displaylink-debian`.](https://github.com/AdnanHodzic/displaylink-debian#how-to-run-displaylink-debian). After you run `displaylink-debian` if you read anything that's printed on screen you should be able to follow simple instructions on how to install/reinstall/uninstall.

Also [read this and open a new issue](https://github.com/AdnanHodzic/displaylink-debian#post-installation-guide-and-troubleshooting) explaining what your problem is along with debug information, and make a reference to this issue.