AdnanHodzic/displaylink-debian #138: connect on boot?

pjlbyrne, 2018-07-06:
Hi, It didn't work on debian stable yesterday, but I installed debian testing (sid) and it did work. My problem is that I want to use this on an intel nuc plugged into a dell 3100 dock. Obviously since it is not a laptop there is no screen to boot on if all screens are on the dock.

I would like both monitors to be enabled via the displaylink driver when the nuc boots. At the moment, I have to plug one monitor into the nuc and a second one into the dock. The 2nd monitor is not recognised until I login to the xfce desktop, open a terminal, and do 'xrand --setprovidersource 1 0'. Then the monitor appears in the display configuration window and can be selected & positioned.

Can I put in some startup configuration so that the displaylink driver will be activated before I login to the desktop? I guess that that would involve running 'xrandr ...' in a script before the greeter runs? Can someone please advise. Thanks.

x10an14, 2018-07-20:
@pjlbyrne Figure out which [display-manager](https://wiki.debian.org/DisplayManager) your Debian install uses (I think it uses [lightdm](https://packages.debian.org/sid/lightdm)?), and then it's a matter of configuring your display-manager to run the `xrandr` commands at startup ([example for lightdm](https://askubuntu.com/a/69501/817936)).

x10an14, 2018-07-22:
I'd be happy to try to offer more suggestions, but then I'd need you to offer more technical details of what you tried, what you expected to happen, and what happened instead.

AdnanHodzic, 2018-11-24:
Please [see following how to resolve](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-configuration) your problem.

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic