AdnanHodzic/displaylink-debian #139: Virtual machine and green screen

moozer, 2018-07-15:
I'm working on using a DL-165 USB device as a physical device for my KVM virtual machine. The "builtin" is a qxl virtual graphics card. 

I am able to run the installation with no error, and I see that the driver is loaded.

```
# lsmod | grep udl
udl                    36864  0
udlfb                  28672  0
drm_kms_helper        196608  3 udl,qxl,evdi
drm                   458752  8 udl,qxl,evdi,ttm,drm_kms_helper
usbcore               290816  7 udl,usbhid,snd_usb_audio,udlfb,xhci_pci,snd_usbmidi_lib,xhci_hcd
```

and the framebuffer device is present on the system
```
# ls /dev/fb* -al
crw-rw---- 1 root video 29, 0 Jul 15 22:37 /dev/fb0
crw-rw---- 1 root video 29, 1 Jul 15 22:37 /dev/fb1
```

X sees it at boot time
```
# cat /var/log/Xorg.0.log | grep -i display
[     3.683] (**) |   |-->Device "DisplayLink"
[     3.698] (II) modeset(0): Creating default Display subsection in Screen section
```

using xrandr (as a user) I get
```
# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x47 cap: 0x2, Sink Output crtcs: 4 outputs: 4 associated providers: 0 name:modesetting

# xrandr -q
Screen 0: minimum 320 x 200, current 1024 x 768, maximum 8192 x 8192
Virtual-1 connected 1024x768+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1024x768      59.95*+
   1920x1200     59.95  
   1920x1080     60.00  
   1600x1200     59.95  
   1680x1050     60.00  
   1400x1050     60.00  
   1280x1024     59.95  
   1440x900      59.99  
   1280x960      59.99  
   1280x854      59.95  
   1280x800      59.96  
   1280x720      59.97  
   1152x768      59.95  
   800x600       59.96  
   848x480       59.94  
   720x480       59.94  
   640x480       59.94  
Virtual-2 disconnected (normal left inverted right x axis y axis)
Virtual-3 disconnected (normal left inverted right x axis y axis)
```

So the `xrandr --setprovideroutputsource 1 0` doesn't work since I don't have the second provider.

I get a "green screen" issue and going through the [issues](https://github.com/AdnanHodzic/displaylink-debian/issues?utf8=%E2%9C%93&q=is%3Aissue+green) I find little help.
Except something about "/etc/modules-load.d/udl.conf", which is not present on my system

The system is a debian/buster with a very recent "apt-get dist-upgrade" and the latest "displaylink-debian".
```
# uname -a
Linux ws-hw 4.16.0-2-amd64 #1 SMP Debian 4.16.16-2 (2018-06-22) x86_64 GNU/Linux
```

This seems to be an X issue. And I have tried [this](https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-375102092) without success.

Your script *just worked* on my other display link device - so I expect there to be something I overlook.
Any suggestions are welcome.

moozer, 2018-07-15:
progress...

grub detects the displaylink, and writes two lines on it, and then X takes over and sometimes I get green screen, but most of the time I get black screen with some grub text.

I am able to put an image on the displaylink using direct framebuffer tools, e.g. `fim /home/moz/Downloads/KittenProgression-Darling-week3.jpg -T 2` - so hardware and drivers are working.

We are back at it being an X issue.

I found the error. You need to blacklist "udlfb", ie. create a file with this content:
```
$ cat /etc/modprobe.d/displaylink.conf 
blacklist udlfb
```

This is resolved now. Sorry for the noise, hopefully someone can use my notes in their debugging.

closed 2018-07-15 by moozer

kaangoksal, 2022-02-06:
I'm trying to do something similar, however in my case qxl has no source output. Can you share your VM configuration?

moozer, 2022-02-06:
> I'm trying to do something similar, however in my case qxl has no source output. Can you share your VM configuration?
> 
Sorry, It is not a system that I have up and running at the moment, so I don't have a config to share.