AdnanHodzic/displaylink-debian #141: Unable to locate package linux-headers-4.9.0-xilinx-v2017.2 file

andy91129, 2018-07-30:
Cannot install Displaylink Driver MPSoC (ubuntu linux 15.04).

E: Unable to locate package linux-headers-4.9.0-xilinx-v2017.2
E: Could not find any package by glob linux-headers-4.9.0-xilinx-v2017.2
E: Could not find any package by regex linux-headers-4.9.0-xilinx-v2017.2
linux-headers-4.9.0-xilinx-v2017.2 installation failed.

Have already tried
sudo apt-get install linux-headers-$(uname -r)

Also, does this driver support ARM platform?

Any assistance would be greatly appreciated.

AdnanHodzic, 2018-07-31:
It does support it, but I believe your kernel is what's preventing it from being installed. 

Could you please return the output of:
`lsb_release -a && uname -r`

Also following stories could be related and might help:
https://github.com/AdnanHodzic/displaylink-debian/issues/67
https://github.com/AdnanHodzic/displaylink-debian/issues/24
https://github.com/AdnanHodzic/displaylink-debian/issues/105
https://github.com/AdnanHodzic/displaylink-debian/issues/103

AdnanHodzic, 2018-11-24:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic

sticky99, 2019-02-04:
Doesn't install correctly on Deepin Linux 15.9, looks to be looking for the linux-headers-4.15.0-29deepin-generic-deepin package rather than the linux-headers-4.15.0-29deepin-generic

`stevo@DeepinLinux:~$ chmod +x displaylink-debian.sh && sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
dpkg-query: package 'linux-headers-4.15.0-29deepin-generic-deepin' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-headers-4.15.0-29deepin-generic-deepin not found! Install? [Y/n] Y 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.15.0-29deepin-generic-deepin
E: Couldn't find any package by glob 'linux-headers-4.15.0-29deepin-generic-deepin'
E: Couldn't find any package by regex 'linux-headers-4.15.0-29deepin-generic-deepin'
linux-headers-4.15.0-29deepin-generic-deepin installation failed.  Aborting.
`

`stevo@DeepinLinux:~$ uname -r
4.15.0-29deepin-generic
stevo@DeepinLinux:~$ sudo apt search linux-headers-$(uname -r)
Sorting... Done
Full Text Search... Done
linux-headers-4.15.0-29deepin-generic/panda,now 4.15.0-29deepin.31 amd64 [installed]
  Linux kernel headers for version 4.15.0 on 64 bit x86 SMP

stevo@DeepinLinux:~$ sudo apt-get install linux-headers-4.15.0-29deepin-generic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-4.15.0-29deepin-generic is already the newest version (4.15.0-29deepin.31).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

AdnanHodzic, 2019-02-20:
I've just merged `displaylink-5` branch with `master` which introduces big changes, among which are support for yesterday released DisplayLink 5.1 driver. Disabling modesetting as part of `displaylink-debian` as now all of it should be happening on EVDI level. I've included few more changes.

Regardless, please pull latest (`git pull`) changes on `master` and run `sudo ./displaylink-debian.sh --install`

If the issue still persists, re-open the issue and include output of `sudo ./displaylink-debian.sh --debug` command.

adamth0mps0n, 2019-07-29:
Doesn't work on a clean install of Deepin.. I get the following error.

sna@sna-PC:~/displaylink-debian$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Deepin
Release: stable
Kernel: 4.15.0-30deepin-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.19.6-3deepin
X11 configs: /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

adamth0mps0n, 2019-07-29:
The latest is Deepin headers are:  linux-headers-4.15.0-30deepin-generic

AceEagle, 2021-03-25:
Doesnt work on a brand new install of kali linux, tried the advices in this issue and others related.

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.10.0-kali3-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: 630
RTX
VGA: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
NVIDIA Corporation TU106BM [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.10-3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x43 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------