AdnanHodzic/displaylink-debian #144: Failing just before the finish line

ChrisM62, 2018-08-03:
This whole issue might be something fairly straightforward, but this part of the system is not one I know well.
I ran the script, it detected my Debian 10 well, reported no errors. 
I went into the post-install section and
1. display is detected (the service is runnning, and I see the displaylink adapter show in the listproviders)
2. xrandr command brings the screen into view for the MATE display tool (I run the MATE 1.20 desktop)
3. When I say 'apply' (after switching it on as the secondary), the system just falls back to the login (LightDM). 

I read somewhere MATE might be problem, how to do I determine this? MATE and dual monitors was not any problem on my previous, intel only laptop. Is it something in the X-server setup? I used to have bumblee and nvidia drivers, (I have a Lenovo P50), but that was no joy. Running NVidia on it's own was also no joy, so I removed all of that and have only the intel video stuff installed. No specific X configs around. 

What do I need to check? I took a look in /var/log, but there was nothing in displaylink, and I couldn't find any mention of displaylink in the logs.

AdnanHodzic, 2018-08-03:
You can always install "gnome-control-center-display" or use some of the other tools as described in [Post installation guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

ChrisM62, 2018-08-03:
I have, MATE has such a tool (mate-display-properties), which is a port of the GNome2 one. After the xrandr, the attached display shows up there. The screen is correctly labeled (make etc...). I select ' active' and 'apply' and after a while I am back at the login prompt. I also tried ARandr, same result.

zachzundel, 2018-08-17:
This might be an issue with xserver 1.2.0, I am trying to figure out similar behavior on Arch.

AdnanHodzic, 2018-11-24:
Could you please tell me if you use any custom X configs, and can you you give me back output of these two commands:

`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`

I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic

ChrisM62, 2018-11-25:
I moved from Debian 10 back to Debian 9, converted that to LMDE 3 and it works perfectly. So in my case I would say something was not as it should be in Debian 10. I don't use custom X configs, if possible. The driver loaded fine on 10, except when told to activate the external display. Then the session bombed. Under Debian 9 however, it works very nicely. All I did was tell the script to assume Debian 9 (as LMDE 3 wasn't in the list, but seeing it uses many of the Debian 9 repos, seemed like a good idea).