AdnanHodzic/displaylink-debian #146: SystemV Support

gagarwa, 2018-08-06:
Currently, on a systemV init, the displaylink driver fails to run with an error, largely due to the service architecture differences between SysV and systemd.  The changes to allow SysV support require changes in the original displaylink adapter.  If anyone knows how, or can guide me, it would be much appreciated.

gagarwa, 2018-08-06:
I will check when I can.

P.S. Sorry, that part in your original message seems to have been cut off on my phone.

gagarwa, 2018-08-11:
I finally got a monitor to test it ..but it failed.  It doesn't create dlm service.  In the beginning, it also gives error, saying init option "-- -" doesn't exist.  I can post additional logs if needed.

Note, changes I made:

1. Added MX distro everywhere Debian was mentioned.  That is in two places, since MX is based on Debian.

2. Added "init" initialization in sysinitdaemon method.

gagarwa, 2018-08-12:
If this is a complex issue (I understand that SysV doesn't even have systemctrl), please let me know so I can move to another distro.

AdnanHodzic, 2018-08-17:
@gagarwa as I said, either switch to any (supported) Debian derived distro with `systemd` or `upstart`, and it should work.

Otherwise, submit a "feature request: sysv support" under issues and link it to this issue. And I can look into it when I have time.

AdnanHodzic, 2018-11-24:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic