AdnanHodzic/displaylink-debian #148: Debian Stretch: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory

sgiebels, 2018-08-06:
..
Performing post install steps

Created symlink /etc/systemd/system/multi-user.target.wants/dlm.service -> /lib/systemd/system/dlm.service.
Disabling PageFlip for modesetting
displaylink-debian.sh: line 285: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
chown: cannot access '/etc/X11/xorg.conf.d/20-displaylink.conf': No such file or directory
chmod: cannot access '/etc/X11/xorg.conf.d/20-displaylink.conf': No such file or directory

# uname -a
Linux ... 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux

AdnanHodzic, 2018-08-06:
Seems to be caused by merging: https://github.com/AdnanHodzic/displaylink-debian/pull/145

pedrib, 2018-08-06:
That's weird @sgiebels you don't have a /etc/X11/xorg.conf.d directory?

sgiebels, 2018-08-06:
$ cat /etc/issue
Debian GNU/Linux 9 \n \l

sgiebels, 2018-08-06:
$ cd /etc/X11
$ ls
Xreset    Xresources  Xsession.d        XvMCConfig       app-defaults             fonts        ja_JP.eucJP  xinit  xsm
Xreset.d  Xsession    Xsession.options  Xwrapper.config  default-display-manager  ja_JP.UTF-8  rgb.txt      xkb

pedrib, 2018-08-06:
can you add
mkdir -p /etc/X11/xorg.conf.d/
before line 285 and see if it works?

sgiebels, 2018-08-06:
I've tried uninstall and re-install with 
git checkout 4a2394bb63ae3f1833aa62c6a843ead54416dc77
This gives:

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory
Disabling PageFlip for modesetting

<rebooting now, will try mkdir afterwards>

AdnanHodzic, 2018-08-06:
@pedrib I believe that's the problem, since `/etc/X11/xorg.conf.d` directory is missing, I just verified this on Ubuntu 18.04 as well.

AdnanHodzic, 2018-08-06:
@sgiebels I just pushed the fix, please `git pull` and run `displaylink-debian` again

sgiebels, 2018-08-06:
I cleard the /usr/src/displaylink-debian folder, did a new git clone .. and tried a reinstall:

Select a key: [i/u/r/q]: r

Uninstalling ...
./displaylink-debian.sh: line 319: displaylink-installer: command not found

-------------------------------------------------------------------
Performing post install steps
Created symlink /etc/systemd/system/multi-user.target.wants/dlm.service -> /lib/systemd/system/dlm.service.
Disabling PageFlip for modesetting
-------------------------------------------------------------------
Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_4.2.zip" file
Removing redundant: "4.2" directory
-------------------------------------------------------------------
Re-install complete, please reboot to apply the changes
-------------------------------------------------------------------

rebooting now..

AdnanHodzic, 2018-08-06:
@sgiebels I also just fixed the error message for uninstall in latest commit:

`./displaylink-debian.sh: line 319: displaylink-installer: command not found` won't happen again with latest change.

sgiebels, 2018-08-06:
Thanks for your quick replies.

That seems to solve the issues of the installer.

Now I'm working on getting my monitor working:
# arandr
<the device DVI-I-1-1 is recognized, with a resolution of  3840x2160, which is correct.>
however, when I enable it, and press apply, I get:

XRandR failed:
XRandR returned error code 1: xrandr: Configure crtc 3 failed

I'll be trying calling xrandr directly again, or might open a new ticket after reading the post-install instructions again.

debug info:
# xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x84 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 1 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

# xrandr --setprovideroutputsource 0 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x84
  Serial number of failed request:  16
  Current serial number in output stream:  17

# xrandr --setprovideroutputsource 1 0

sgiebels, 2018-08-06:
closing as original installer issue has been resolved. thanks again.

closed 2018-08-06 by sgiebels