AdnanHodzic/displaylink-debian #151: Not working?

DayehT5N, 2018-08-10:
I installed displaylink-debian successfully in kali linux.
I run on the terminal:
``` xrandr --listproviders
 ```
And it says there are 3 providers.

i run ```
xrandr --setprovideroutputsource 1 0
```
then i go to display settings and see that my moniter appeared and i put it and then i apply the changes, then the os automatically restarts and i log in and the moniter did not work and i go to settings and i dont find my moniter there.
i am using a usb adapter to plug in my moniter.
Help?

gagarwa, 2018-08-14:
As soon as you input that command it should work.  If you restart, it will reset, and you have to redo command.  To make it work permanently, you have to add it to init or when you start bash or something similar.

To extend desktop, try the command line commands, rather than the display sertings so it doesn't restart.

DayehT5N, 2018-08-15:
Thats the thing, when i put that command, and press enter, it restarts automatically, i did not want it to restart, like as soon as i input it it restats it gives me the login screen, i dont know what to do..

gagarwa, 2018-08-15:
Bash is the most common Unix shell.  When your start terminal, it starts bash shell.  When it does, it initializes using commands in ./bashrc.  Many systems start bash on system start also, so it is a easy way to run commands on start.

Init or init.d is a folder in your system. It contains scripts that are run during system initialization or startup.

Note: I am not an expert either, so if I made a please, please correct me.

DayehT5N, 2018-08-15:
aha ok, so what can i exactly do to fix my problem? :|

gagarwa, 2018-08-15:
It is restarting on this command? ` xrandr --output DVI-1-0 --auto --right-of eDP1`
If so, try changing the init settings.  See [Kali Forums](https://forums.kali.org/showthread.php?19841-Help-with-Screen-Resolution), and [Stack Exchange](https://unix.stackexchange.com/questions/331398/starting-script-to-set-monitor-configuration)

If it is something else, look at the [post install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) again, and particularly at xrandr.

**You may just want to look more into xrandr, to see if you can find similar commands.  I would try this first, to make sure that you don't change something in the init when you don't know what you are doing.**