AdnanHodzic/displaylink-debian #153: No providers listed

Azza007, 2018-08-16:
Hey,

So I installed the driver on Ubuntu 18.04 and it is active and running. When I plug my dock in my laptop screen flashes as if a display has been connected, however when I run the command xrandr --listproviders it shows 0 providers. When I also run "lsusb -d 17e9:" it shows 2 entries, both are my dock.

Any help would be much apricated.

AdnanHodzic, 2018-08-17:
Have you looked at [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md), in particular making sure secure boot is disabled.

Azza007, 2018-08-17:
Hey,

So I actually fixed it. I reinstalled Ubuntu 18.04 and the driver, and it started working. Must have been something wrong in my install.

AdnanHodzic, 2018-08-19:
Great, glad you were able to resolve the issue.

closed 2018-08-19 by AdnanHodzic