AdnanHodzic/displaylink-debian #156: No monitors Detected on Kubuntu (ubuntu) 18.4

awsaunier, 2018-09-12:
A few days ago from writing I installed kubuntu on my computer with a dual boot to windows 10.  I ran the Displaylink driver installer and was not able to complete install. I removed what was installed by the driver and then I ran your script and had no problems installing.  I followed your post installation guide and I cannot get the service to start.  When I start the service I get the following error: 
`asuanier@asaunier-Linux:~$ systemctl start dlm.service`
`Failed to start dlm.service: Unit dlm.service not found.`
`asuanier@asaunier-Linux:~$ `

This is more information on my system:
`Kubuntu 18.04`
`KDE plasma Version: 5.12.6`
`KDE Frameworks Version: 5.44.0`
`Qt Version: 5.9.5`
`Kernel Version: 4.15.0-34-generic`
`OS Type: 64-bit`

Hardware
`CPU: Intel 3200u`
`GPU: HD 520`
`RAM: 8GB`

Thank you for all you do and taking time to work on this project.

AdnanHodzic, 2018-09-12:
Just to make sure, have you disabled UEFI as it's suggested in [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

awsaunier, 2018-09-13:
That serves me right for not reading the post first then trying the commands.  Now that i have UEFI off the service is working properly but a new issue has appeared.  I have an intel gpu and I tried the solution suggested in #68 and that helped the rendering problem.  Finally I enabled the displays and now it is all working.  Thank you for taking the time to help me.

closed 2018-09-13 by awsaunier