AdnanHodzic/displaylink-debian #160: xrandr --setprovideroutputsource 1 0 failed to run

zioalex, 2018-09-26:
Hi guys trying to get the Dell displaylink hub on my laptop but I have some problem.
I have a Sabayon machine.
I managed to install the proper kernel and displaylink package.
Some info:
* Linux sabayon 4.18.5-pentoo
* evdi-1.5.0-r2
* x11-drivers/displaylink-driver-4.1.9
* x11-base/xorg-server-1.19.5-r2

Noticed that when the evdi module is loaded I get the follow message:
`
17.592680] evdi: loading out-of-tree module taints kernel.
[   17.592772] evdi: no symbol version for drm_framebuffer_cleanup
[   17.593774] evdi: [I] Initialising logging on level 5
[   17.593776] evdi: [I] Atomic driver:no
`
Not sure if it means something useful.

I tried with both :
::::::::::::::
20-displaylink.conf1
::::::::::::::
Section "Device" 
  Identifier "Intel Graphics"
  Driver "intel"
EndSection

::::::::::::::
20-displaylink.conf2
::::::::::::::
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection 

As suggested here https://github.com/AdnanHodzic/displaylink-debian/issues/68

While running with the 20-displaylink.conf1 I could not see 2 providers.
Now I am running with the 20-displaylink.conf2 and I can see the 2 providers:

```
> xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x66 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
```

Then I try to connect the provider and I receive the follow error.
```
> xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  139 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x3f
  Serial number of failed request:  16
  Current serial number in output stream:  17
```
Any suggestions.

bartkdm, 2018-11-01:
Hello, 

I'm experiencing the same problem:

uname -a

> Linux 4.15.0-38-generic #41~16.04.1-Ubuntu SMP Wed Oct 10 20:16:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

xrandr --listproviders

> Providers: number : 2
> Provider 0: id: 0x23c cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
> Provider 1: id: 0x46 cap: 0x2, Sink Output crtcs: 3 outputs: 4 associated providers: 1 name:modesetting

systemctl status dlm.service

> ● dlm.service - DisplayLink Manager Service
>    Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
>    Active: active (running) since do 2018-11-01 14:20:40 CET; 29min ago
>   Process: 1263 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && modprobe evdi) (code=exited, status=0/SUCCESS)
>  Main PID: 1271 (DisplayLinkMana)
>    CGroup: /system.slice/dlm.service
>            └─1271 /opt/displaylink/DisplayLinkManager
> 
> nov 01 14:20:39 systemd[1]: Starting DisplayLink Manager Service...
> nov 01 14:20:40 systemd[1]: Started DisplayLink Manager Service.

xrandr --setprovideroutputsource 1 0

> (no output)

xrandr --setprovideroutputsource 2 0

> (expecting error) Could not find provider with index 2

xrandr --setprovideroutputsource 0 0

> X Error of failed request:  BadValue (integer parameter out of range for operation)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  35 (RRSetProviderOutputSource)
>   Value in failed request:  0x23c
>   Serial number of failed request:  16
>   Current serial number in output stream:  17
> 
> 

Hope somebody can help me out! No screens detected in ubuntu display setting ...

Thanks in advance!
Fred.

baldhead, 2018-11-01:
Hello, 

I'm experiencing the same problem:

uname -a

> Linux 4.15.0-38-generic #41~16.04.1-Ubuntu SMP Wed Oct 10 20:16:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

xrandr --listproviders

> Providers: number : 2
> Provider 0: id: 0x23c cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
> Provider 1: id: 0x46 cap: 0x2, Sink Output crtcs: 3 outputs: 4 associated providers: 1 name:modesetting

systemctl status dlm.service

> ● dlm.service - DisplayLink Manager Service
>    Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
>    Active: active (running) since do 2018-11-01 14:20:40 CET; 29min [...]

xrandr --setprovideroutputsource 1 0

> (no output)

xrandr --setprovideroutputsource 2 0

> (expecting error) Could not find provider with index 2

xrandr --setprovideroutputsource 0 0

> X Error of failed request:  BadValue (integer parameter out of range for operation)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  35 (RRSetProviderOutputSource) [...]

Hope somebody can help me out! No screens detected in ubuntu display setting ...

Thanks in advance!
Fred.

baldhead, 2018-11-30:
> Can both of you please give me the output of following commands:
>
> lspci | grep -i vga
> lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
>
> With latest update, /etc/X11/xorg.conf.d/20-displaylink.conf will be
> populated with
>
> Section "Device"
>   Identifier  "Intel"
>   Driver      "intel"
> EndSection
>
> only if your card is i915

My output:
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
i915

AdnanHodzic, 2018-12-01:
@zioalex what's your output of `lspci | grep -i vga`?

@baldhead you have both Intel and Nvidia cards? What happens if you try what I suggest in: https://github.com/AdnanHodzic/displaylink-debian/issues/120#issuecomment-443434739

zioalex, 2018-12-03:
@AdnanHodzic  this is my VGA card:
`00:02.0 VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
`

AdnanHodzic, 2018-12-03:
@zioalex can you please get me the output of following two commands:

`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'` 
and
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

Can you tell me what are the current contents of your: `/etc/X11/xorg.conf.d/20-displaylink.conf` file?

and if it's not this content already update it to:

```
Section "Device"
  Identifier  "Intel"	  Identifier  "Intel"
  Driver      "intel"	  Driver      "intel"
EndSection	EndSection
```

after which just log out and login again and see if things are working as expected.

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic

rachel-kt, 2019-02-15:
Output for `_cat /etc/X11/xorg.conf.d/20-displaylink.conf_`
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

Output for `_lspci | grep -i vga_`
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
01:00.0 VGA compatible controller: NVIDIA Corporation GF108M [GeForce GT 525M] (rev a1)

`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
i915

`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`
Core
GT

Thanks for the help!

AdnanHodzic, 2019-02-16:
@rachel-kt it's really hard for me to follow what's happening as part of this issue. Could you create a new issue and include output of `sudo ./displaylink-debian.sh --debug` command along with it.