AdnanHodzic/displaylink-debian #162: Laptop Radeon hybrid graphics

hadogenes, 2018-10-08:
Hi,
how to set _amdgpu_ driver for my radeon graphics, because when I have _/usr/share/X11/xorg.conf.d/20-displaylink.conf_ with 
```
Section "Device"
  Identifier "Intel"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```
it sets me modesetting for both graphics cards

AdnanHodzic, 2018-11-24:
I would like to extend support for your graphics card. But before I could do that could you please give me the output of following commands:

`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic