AdnanHodzic/displaylink-debian #163: linux-headers-3.18.41+ installation failed

geminia, 2018-10-09:
Hi have a Gemini PDA - Mediatek processor, running Debian.  

Checking dependencies

unzip is installed
dpkg-query: package 'linux-headers-3.18.41+' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-headers-3.18.41+ not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree      
Reading state information... Done
E: Unable to locate package linux-headers-3.18.41
E: Couldn't find any package by glob 'linux-headers-3.18.41'
E: Couldn't find any package by regex 'linux-headers-3.18.41'
linux-headers-3.18.41+ installation failed.  Aborting.

After looking at other issues I have also run this check

gemini@localhost:/$ lsb_release -a && uname -r
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 9.5 (stretch)
Release:        9.5
Codename:       stretch
3.18.41+

The kernel source code for the device is here
https://github.com/gemian/gemini-linux-kernel-3.18

redjamesg, 2018-10-12:
What does the command

'dpkg -l | grep linux-headers' return?
What is present in /lib/modules ?

geminia, 2018-10-15:
> What does the command
> 
> 'dpkg -l | grep linux-headers' return?
> What is present in /lib/modules ?

ok so first first cmd returned nothing, so I ran it for grep linux. Here is the result of that.

> gemini@localhost:~$ sudo dpkg -l | grep linux
> ii  libselinux1:arm64 
> 2.6-3+b3                                          arm64 SELinux 
> runtime shared libraries
> ii  libv4l-0:arm64 1.12.3-1                                          
> arm64 Collection of video4linux support libraries
> ii  libv4lconvert0:arm64 
> 1.12.3-1                                          arm64 Video4linux 
> frame format conversion library
> ii  util-linux 2.29.2-1+deb9u1                                   arm64 
> miscellaneous system utilities

I then tried running this cmd ' sudo apt-get install linux-headers-$(uname -r) '

> gemini@localhost:/lib$ sudo apt-get install linux-headers-$(uname -r)
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package linux-headers-3.18.41
> E: Couldn't find any package by glob 'linux-headers-3.18.41'
> E: Couldn't find any package by regex 'linux-headers-3.18.41'

I could not find lib/module folder but here is what I think might be the equivalent.

> gemini@localhost:/lib/aarch64-linux-gnu$ ls
> ld-2.24.so               libdns-export.so.162 
> libmnl.so.0                libpopt.so.0
> ld-linux-aarch64.so.1    libdns-export.so.162.1.3 
> libmnl.so.0.2.0            libpopt.so.0.0.0
> libBrokenLocale-2.24.so  libe2p.so.2 libmount.so.1              
> libprocps.so.6
> libBrokenLocale.so.1     libe2p.so.2.3 libmount.so.1.1.0          
> libprocps.so.6.0.0
> libSegFault.so           libexpat.so.1 libncurses.so.5            
> libpthread-2.24.so
> libacl.so.1              libexpat.so.1.6.2 libncurses.so.5.9          
> libpthread.so.0
> libacl.so.1.1.0          libext2fs.so.2 libncursesw.so.5           
> libreadline.so.5
> libaio.so.1              libext2fs.so.2.4 libncursesw.so.5.9         
> libreadline.so.5.2
> libaio.so.1.0.1          libfdisk.so.1 libnl-3.so.200             
> libreadline.so.7
> libanl-2.24.so           libfdisk.so.1.1.0 libnl-3.so.200.22.0        
> libreadline.so.7.0
> libanl.so.1              libfuse.so.2 libnl-genl-3.so.200        
> libresolv-2.24.so
> libapparmor.so.1         libfuse.so.2.9.7 libnl-genl-3.so.200.22.0   
> libresolv.so.2
> libapparmor.so.1.4.0     libgcc_s.so.1 libnsl-2.24.so             
> librt-2.24.so
> libatasmart.so.4         libgcrypt.so.20 libnsl.so.1                
> librt.so.1
> libatasmart.so.4.0.5     libgcrypt.so.20.1.6 
> libnss_compat-2.24.so      libseccomp.so.2
> libattr.so.1             libglib-2.0.so.0 libnss_compat.so.2         
> libseccomp.so.2.3.1
> libattr.so.1.1.0         libglib-2.0.so.0.5000.3 
> libnss_dns-2.24.so         libselinux.so.1
> libaudit.so.1            libgpg-error.so.0 libnss_dns.so.2            
> libsepol.so.1
> libaudit.so.1.0.0        libgpg-error.so.0.21.0 
> libnss_files-2.24.so       libslang.so.2
> libblkid.so.1            libhistory.so.5 libnss_files.so.2          
> libslang.so.2.3.1
> libblkid.so.1.1.0        libhistory.so.5.2 libnss_hesiod-2.24.so      
> libsmartcols.so.1
> libbsd.so.0              libhistory.so.7 libnss_hesiod.so.2         
> libsmartcols.so.1.1.0
> libbsd.so.0.8.3          libhistory.so.7.0 libnss_mdns.so.2           
> libss.so.2
> libbz2.so.1              libidn.so.11 libnss_mdns4.so.2          
> libss.so.2.0
> libbz2.so.1.0            libidn.so.11.6.16 libnss_mdns4_minimal.so.2  
> libsystemd.so.0
> libbz2.so.1.0.4          libirs-export.so.141 
> libnss_mdns6.so.2          libsystemd.so.0.17.0
> libc-2.24.so             libirs-export.so.141.0.4 
> libnss_mdns6_minimal.so.2  libthread_db-1.0.so
> libc.so.6                libisc-export.so.160 
> libnss_mdns_minimal.so.2   libthread_db.so.1
> libcap-ng.so.0           libisc-export.so.160.0.0 
> libnss_nis-2.24.so         libtinfo.so.5
> libcap-ng.so.0.0.0       libisccc-export.so.140 
> libnss_nis.so.2            libtinfo.so.5.9
> libcap.so.2              libisccc-export.so.140.0.4 
> libnss_nisplus-2.24.so     libudev.so.1
> libcap.so.2.25           libisccfg-export.so.140 
> libnss_nisplus.so.2        libudev.so.1.6.5
> libcidn-2.24.so          libisccfg-export.so.140.3.0 
> libntfs-3g.so.871          libulockmgr.so.1
> libcidn.so.1             libiw.so.30 libntfs-3g.so.871.0.0      
> libulockmgr.so.1.0.1
> libcom_err.so.2          libkeyutils.so.1 libpam.so.0                
> libusb-1.0.so.0
> libcom_err.so.2.1        libkeyutils.so.1.5 libpam.so.0.83.1           
> libusb-1.0.so.0.1.0
> libcrypt-2.24.so         libkmod.so.2 libpam_misc.so.0           
> libutil-2.24.so
> libcrypt.so.1            libkmod.so.2.3.1 libpam_misc.so.0.82.0      
> libutil.so.1
> libcryptsetup.so.4       liblzma.so.5 libpamc.so.0               
> libuuid.so.1
> libcryptsetup.so.4.7.0   liblzma.so.5.2.2 libpamc.so.0.82.1          
> libuuid.so.1.3.0
> libdbus-1.so.3           liblzo2.so.2 libparted.so.2             
> libwrap.so.0
> libdbus-1.so.3.14.15     liblzo2.so.2.0.0 libparted.so.2.0.1         
> libwrap.so.0.7.6
> libdevmapper.so.1.02.1   libm-2.24.so libpcprofile.so            libz.so.1
> libdl-2.24.so            libm.so.6 libpcre.so.3               
> libz.so.1.2.8
> libdl.so.2               libmemusage.so libpcre.so.3.13.3          
> security

Thanks for helping out

redjamesg, 2018-10-15:
What about the command 'apt search linux-headers' see if there is something like linux-headers-common or similar.

geminia, 2018-10-15:
Here is the result for apt search linux-headers

 Sorting... Done
> Full Text Search... Done
> aufs-dkms/stable 4.9+20161219-1 arm64
>   DKMS files to build and install aufs
>
> linux-headers-4.17.0-0.bpo.1-all/stretch-backports 4.17.8-1~bpo9+1 arm64
>   All header files for Linux 4.17 (meta-package)
>
> linux-headers-4.17.0-0.bpo.1-all-arm64/stretch-backports 
> 4.17.8-1~bpo9+1 arm64
>   All header files for Linux 4.17 (meta-package)
>
> linux-headers-4.17.0-0.bpo.1-arm64/stretch-backports 4.17.8-1~bpo9+1 arm64
>   Header files for Linux 4.17.0-0.bpo.1-arm64
>
> linux-headers-4.17.0-0.bpo.1-common/stretch-backports 4.17.8-1~bpo9+1 all
>   Common header files for Linux 4.17.0-0.bpo.1
>
> linux-headers-4.17.0-0.bpo.3-all/stretch-backports 4.17.17-1~bpo9+1 arm64
>   All header files for Linux 4.17 (meta-package)
>
> linux-headers-4.17.0-0.bpo.3-all-arm64/stretch-backports 
> 4.17.17-1~bpo9+1 arm64
>   All header files for Linux 4.17 (meta-package)
>
> linux-headers-4.17.0-0.bpo.3-arm64/stretch-backports 4.17.17-1~bpo9+1 
> arm64
>   Header files for Linux 4.17.0-0.bpo.3-arm64
>
> linux-headers-4.17.0-0.bpo.3-common/stretch-backports 4.17.17-1~bpo9+1 all
>   Common header files for Linux 4.17.0-0.bpo.3
>
> linux-headers-4.18.0-0.bpo.1-all/stretch-backports 4.18.6-1~bpo9+1 arm64
>   All header files for Linux 4.18 (meta-package)
>
> linux-headers-4.18.0-0.bpo.1-all-arm64/stretch-backports 
> 4.18.6-1~bpo9+1 arm64
>   All header files for Linux 4.18 (meta-package)
>
> linux-headers-4.18.0-0.bpo.1-arm64/stretch-backports 4.18.6-1~bpo9+1 arm64
>   Header files for Linux 4.18.0-0.bpo.1-arm64
>
> linux-headers-4.18.0-0.bpo.1-common/stretch-backports 4.18.6-1~bpo9+1 all
>   Common header files for Linux 4.18.0-0.bpo.1
>
> linux-headers-4.18.0-0.bpo.1-common-rt/stretch-backports 
> 4.18.6-1~bpo9+1 all
>   Common header files for Linux 4.18.0-0.bpo.1-rt
>
> linux-headers-4.18.0-0.bpo.1-rt-arm64/stretch-backports 
> 4.18.6-1~bpo9+1 arm64
>   Header files for Linux 4.18.0-0.bpo.1-rt-arm64
>
> linux-headers-4.9.0-4-common-grsec/stretch-backports 
> 4.9.65-2+grsecunoff1~bpo9+1 all
>   Common header files for Linux 4.9.0-4-grsec
>
> linux-headers-4.9.0-7-all/stable 4.9.110-1 arm64
>   All header files for Linux 4.9 (meta-package)
>
> linux-headers-4.9.0-7-all-arm64/stable 4.9.110-1 arm64
>   All header files for Linux 4.9 (meta-package)
>
> linux-headers-4.9.0-7-arm64/stable 4.9.110-1 arm64
>   Header files for Linux 4.9.0-7-arm64
>
> linux-headers-4.9.0-7-common/stable 4.9.110-1 all
>   Common header files for Linux 4.9.0-7
>
> linux-headers-4.9.0-7-common-rt/stable 4.9.110-1 all
>   Common header files for Linux 4.9.0-7-rt
>
> linux-headers-arm64/stable 4.9+80+deb9u5 arm64
>   Header files for Linux arm64 configuration (meta-package)
>
> linux-headers-rt-arm64/stretch-backports 4.18+98~bpo9+1 arm64
>   Header files for Linux rt-arm64 configuration (meta-package)
>
> linux-libc-dev-alpha-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-arm64-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-armel-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-armhf-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-hppa-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-m68k-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-mips-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-mips64-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-mips64el-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-mipsel-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-powerpc-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-powerpcspe-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-ppc64-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-ppc64el-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-s390x-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-sh4-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)
>
> linux-libc-dev-sparc64-cross/stable 4.9.25-1cross1 all
>   Linux Kernel Headers for development (for cross-compiling)

geminia, 2018-10-28:
Hi @redjamesg does the last update shed any light on it? thanks

AdnanHodzic, 2018-11-24:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-11-24 by AdnanHodzic