AdnanHodzic/displaylink-debian #164: Night Light

jeffbeagley, 2018-10-16:
Anyone else have issues with nightlight? It appears to be working on my laptop screen but not my two externals. I have updated to the latest 4.2 drivers

AdnanHodzic, 2018-11-24:
Most welcome :)

It seems to me like this is a GNOME (or whatever) desktop environment you're using, so if I was you I would try following up on this problem with them.

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic

jeffbeagley, 2018-12-04:
Install was successful, however its still only adjusting on my laptop. I am using the EVGA UV Plus+39 external card for my external monitors

cat /etc/X11/xorg.conf.d/20-displaylink.conf

```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

lspci | grep -i vga
```
00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)

```

lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
```
i915
```
lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
```
Core
```