AdnanHodzic/displaylink-debian #165: Using ARANDR and applying the config causes Kali to log out

MatthewHarrold, 2018-10-16:
I can configure the screens as they are recognised when I use :

xrandr --listproviders

AdnanHodzic, 2018-11-24:
First can you please verify that you went through [post installation guide](https://github.com/AdnanHodzic/displaylink-debian)? 

If you have, can you please confirm if you have the same problem after pulling latest updates and running reinstall?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic