AdnanHodzic/displaylink-debian #168: Ethernet not detected.

chrisorcox, 2018-10-24:
I'm not sure if anyone else has had issues with ethernet with this driver (specifically on Ubuntu 18.04). My computer refused to detect that an ethernet cable was plugged into the displaylink dock. Even more interestingly, when I would unplug the ethernet cable, my computer would suddenly detect that an ethernet cable was plugged in (but it wasn't) and try to connect for a bit before failing. If I plugged the ethernet back in, it would automatically go back to not detecting an ethernet cable. For awhile, I thought that this was an issue with the dock and not this driver. However, recently, I reinstalled the OS and plugged in the dock without having these drivers, and the ethernet shows up! Of course, I can't use the hdmi connection without these drivers, but the ethernet is back! As a test, I tried re-installing the drivers, and again, I get the same ethernet issues. Uninstall the driver, and the ethernet works again...

AdnanHodzic, 2018-11-24:
That's a very weird issue, and I've never came across one such issue, try submitting or looking for similar issue [on upstream project](https://github.com/DisplayLink/evdi/issues)

AdnanHodzic, 2018-12-04:
I'm closing this issue for inactivity. 

Please re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2018-12-04 by AdnanHodzic