AdnanHodzic/displaylink-debian #169: Doesn't work on Kali 2018.3

vampireactual, 2018-10-27:
Same issue as #163, but kali doesn't have information for linux-header-4.17.0-kali1-amd64. Any solution?

Output of install script:
-------------------------------------------------------------------

Checking dependencies

unzip is installed
dpkg-query: package 'linux-headers-4.17.0-kali1-amd64' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
linux-headers-4.17.0-kali1-amd64 not found! Install? [Y/n] Y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.17.0-kali1-amd64
E: Couldn't find any package by glob 'linux-headers-4.17.0-kali1-amd64'
E: Couldn't find any package by regex 'linux-headers-4.17.0-kali1-amd64'
linux-headers-4.17.0-kali1-amd64 installation failed.  Aborting.

_________________________________________________________________________________

Output lsb_release -a && uname -r:
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	kali-rolling
Codename:	kali-rolling
4.17.0-kali1-amd64

Output of apt-cache search linux-headers-4.*:
linux-headers-4.18.0-kali2-all - All header files for Linux 4.18 (meta-package)
linux-headers-4.18.0-kali2-all-amd64 - All header files for Linux 4.18 (meta-package)
linux-headers-4.18.0-kali2-amd64 - Header files for Linux 4.18.0-kali2-amd64
linux-headers-4.18.0-kali2-cloud-amd64 - Header files for Linux 4.18.0-kali2-cloud-amd64
linux-headers-4.18.0-kali2-common - Common header files for Linux 4.18.0-kali2
linux-headers-4.18.0-kali2-common-rt - Common header files for Linux 4.18.0-kali2-rt
linux-headers-4.18.0-kali2-rt-amd64 - Header files for Linux 4.18.0-kali2-rt-amd64

AdnanHodzic, 2018-11-09:
Please pull the latest changes where I've added support for Kali 2018.3

vampireactual, 2018-11-20:
Is there any way to make it persistent? If I reboot or something then the screen turns green, then I have to unplug the usb adapter, reinstall the drivers, reboot, then plug it back in, only then does it work only about 50% of the time.

AdnanHodzic, 2018-11-24:
@vampireactual first of all please make sure to go through: [post installation guide](https://github.com/AdnanHodzic/displaylink-debian)

I believe the "persistency" problem you're mentioning is described right here: https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-configuration

Once connected it should work all the time, did you try plugging in DisplayLink into different USB port?

vampireactual, 2018-11-24:
I’ve tried both of those steps previously and I followed the post
installation guide. But I will try again.

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic